kliment/Printrun #934: Cannot get connection to the printer with Mariln Firmware

barci2, 2018-04-13:
Hi
I have recently got printrun. However, it does not connect to the printer with a marlin firmware installed on a clone of Arduino Mega. What should I do? I run it obviously with sudo pronterface.py command, connecting by /dev/ttyUSB1 port with baundrate 115200(as I set so in firmware config). Here are outputs of some commands that might be useful:
- ls -lah /dev/ttyUSB1:
crw-rw---- 1 root uucp 188, 1 Apr 13 20:04 /dev/ttyUSB1
- lsusb
Bus 002 Device 003: ID 05e3:0749 Genesys Logic, Inc. 
Bus 002 Device 002: ID 2109:0813 VIA Labs, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 008: ID 058f:6387 Alcor Micro Corp. Flash Drive
Bus 001 Device 007: ID 2109:2813 VIA Labs, Inc. 
Bus 001 Device 006: ID 0bda:58c2 Realtek Semiconductor Corp. 
Bus 001 Device 005: ID 0cf3:e300 Qualcomm Atheros Communications 
Bus 001 Device 004: ID 187c:0530 Alienware Corporation 
Bus 001 Device 010: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
Bus 001 Device 002: ID 1532:004c Razer USA, Ltd 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
- marlin firmware config file
/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

#include "boards.h"
#include "macros.h"

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
[... 1052 lines cut ...]
// Uncomment below to enable
//#define FILAMENT_WIDTH_SENSOR

#define DEFAULT_NOMINAL_FILAMENT_DIA 3.00  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation

#if ENABLED(FILAMENT_WIDTH_SENSOR)
  #define FILAMENT_SENSOR_EXTRUDER_NUM 0   //The number of the extruder that has the filament sensor (0,1,2)
  #define MEASUREMENT_DELAY_CM        14   //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

  #define MEASURED_UPPER_LIMIT         3.30  //upper limit factor used for sensor reading validation in mm
  #define MEASURED_LOWER_LIMIT         1.90  //lower limit factor for sensor reading validation in mm
  #define MAX_MEASUREMENT_DELAY       20     //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

  #define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

  //When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
  //#define FILAMENT_LCD_DISPLAY
#endif

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //CONFIGURATION_H

I would be really thankful for any help.

hurzl, 2018-11-01:
No Poblem here with Marlin 1.1.1
```
Marlin 1.1.1
echo: Last Updated: 2017-05-12 12:00 | Author: (none, default config)
Compiled: Jun 13 2017

```

rockstorm101, 2018-11-02:
Hi @barci2, after @hurzl's comment it seems like either a connection or a pronterface issue. Did you find a solution to your problem? If so, please do close this bug, if not, please show us the output of pronterface when trying to connect.

rockstorm101, 2020-05-23:
Due to the lack of feedback I assume the issue is solved. Please feel free to reopen if needed.

closed 2020-05-23 by rockstorm101