kliment/Printrun #944: Disable file parsing when loading files to lower file loading times to zero

sebascarra, 2018-05-29:
Hello, I use pronsole to print from the command line using a Raspberry Pi. Loading large files takes a couple minutes and honestly I just need to stream the file to another board. I'm not sure on why gcoder needs to run when loading from the command line, I don't need the ETA or anything like that. How can I disable gcoder or whatever is required to reduce loading times to near zero??

Thanks.

XL-Reaper, 2019-05-07:
Same question, I would also like to have the option to disable parsing,

volconst, 2019-06-27:
This could be achievable with 'pronsole> run_gcode_script cat path/to/file.gcode' since commit https://github.com/kliment/Printrun/commit/9298c8dc14acaa4ff90dbdd6c672d7aced8b6d51 .
run_gcode_script now uses do_print, so it should be able to be paused and resumed, but may leak handles if output not exhausted.