kliment/Printrun #947: Exception in thread Thread on macOS High sierra

mosvov, 2018-06-15:
On version 2.0.0rc5 when trying to connect to remote printer (192.168.1.121:23) I get error 
Connecting...
Exception in thread Thread-8:
Traceback (most recent call last):
  File "threading.py", line 916, in _bootstrap_inner
  File "threading.py", line 864, in run
  File "printrun/printcore.py", line 345, in _listen
  File "printrun/printcore.py", line 308, in _listen_until_online
  File "printrun/printcore.py", line 672, in _send
TypeError: write() argument must be str, not bytes

On 1.6.0 with same settings it works good.

tarmof, 2018-11-28:
Yes, same here with latest git pull

Connecting...
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/Printrun/printrun/printcore.py", line 345, in _listen
    self._listen_until_online()
  File "/opt/Printrun/printrun/printcore.py", line 308, in _listen_until_online
    self._send("M105")
  File "/opt/Printrun/printrun/printcore.py", line 672, in _send
    self.printer.write((command + "\n").encode('ascii'))
TypeError: write() argument must be str, not bytes

k3ns, 2018-12-16:
See Issue #936

closed 2024-08-15 by mosvov