kliment/Printrun #954: alternate UI simple like the BIG rep onr has made

amrithmmh, 2018-07-11:
how customisable is the pronterface ui and how to customise it?
has anyone made a custom beatiful simple UI based on pronterface?

amrithmmh, 2018-07-11:
is it possible to use Qt5 to create a alternate UI for pronterface

amrithmmh, 2018-07-12:
or maybe Kivy which is cross platform and looking good too

rockstorm101, 2018-07-12:
Hi @amrithmmh, currently Pronterface is the UI for Pronsole which is a command line 3d printer host. Printcore is the library for creating hosts. If it is just the UI what you want to create, I'd suggest using Pronsole as a "backend".

I was once tried to adapt Pronterface to small screens to fit Raspberry Pis and the like, a "Tabbed" mode was created. However this feature has now been disabled because it needs some rewriting to work properly. You are more than welcome to work on this if you want.