kliment/Printrun #958: BUG: PrintRun fails to display full image

dcwalmsley, 2018-08-01:
Good Day,

I'm currently running Printrun 2014.08.01 version and Ive noticed that small gcode files will display on the plater but larger more complex files receive errors similar to what I pasted below.  Is there a newer version which will run on Windows 10 and display all sized files?

Is there any plans to develop newer versions of this program?  Thanks for all that you do.

R,
Doug W
------------------------------------------------------------------------------------------
Loading file: C:\...\Desktop\loubie_adalinda_dragon - with ooze layer.gcode
Exception in thread Thread-8:
Traceback (most recent call last):
  File "threading.pyc", line 808, in __bootstrap_inner
  File "threading.pyc", line 761, in run
  File "printrun\pronterface.pyc", line 1452, in loadviz
  File "printrun\gcview.pyc", line 338, in addfile_perlayer
  File "printrun\gl\libtatlin\actors.pyc", line 429, in load_data
MemoryError: cannot allocate memory for array

dcwalmsley, 2018-08-01:
So I downloaded and ran the 2017 version and ran into the same problem.

------------------------------------------------------------------------------------------------
Loading file: C:\Users\dcwal\Desktop\loubie_adalinda_dragon - with ooze layer.gcode
Exception in thread Thread-2:
Traceback (most recent call last):
  File "threading.pyc", line 801, in __bootstrap_inner
  File "threading.pyc", line 754, in run
  File "printrun\pronterface.pyc", line 1543, in loadviz
  File "printrun\gcview.pyc", line 338, in addfile_perlayer
  File "printrun\gl\libtatlin\actors.pyc", line 427, in load_data
MemoryError: cannot allocate memory for array

CostanzoPablo, 2020-04-12:
Already fixed. You must put: (i tried on windows, and get working)

git clone
py -m venv venv
venv\Scripts\activate
python -m pip install -r requirements.txt
python pronterface.py