kliment/Printrun #962: problem with installing Python dependencies on Windows

mendelusr, 2018-09-04:
I only want to install Python and its dependencies on Windows.

- I have successfully downloaded Python 3.6.

- I have cloned the repository using
> git clone https://github.com/kliment/Printrun.git

- I have created a virtual environment using
> py -3 -m venv venv

- I have activated the virtual environment using
> venv\Scripts\activate

- I have tried to install the dependencies using
> python -m pip install -r requirements.txt
but I only get "No such file or directory"

- I have searched user\venv for a file "requirements.txt" but only got
C:\Users\user\venv\Lib\site-packages\pip\_vendor\packaging\requirements.py
C:\Users\user\venv\Lib\site-packages\setuptools\_vendor\packaging\requirements.py
C:\Users\user\venv\Lib\site-packages\pkg_resources\_vendor\packaging\requirements.py

Can someone help me find the file required for installing the dependencies?

hroncok, 2018-09-04:
requirements.txt is in the cloned repository. you need to change directory to it (something like `cd Printrun`)

hroncok, 2018-09-04:
I guess taht after

```
$ git clone https://github.com/kliment/Printrun.git  # clone the repository
```

we need to add

```
$ cd Printrun
```

to make the docs accurate

closed 2018-09-12 by kliment