kliment/Printrun #965: Where does M114 output print from printcore call?

KaplanAlex, 2018-09-24:
This is more of a question than an issue, but I've been troubleshooting this for an hour and havent found any information on it. I'm using just the printcore package to control a motor and want to get its position with M114. I've tried using "printer.startprint(['M114'])" and "pos = printer.send_now('M114')" and no errors are thrown, but I have no idea where the result is going. I'm able to move the motor with other GCODE commands so that isn't the issue. Also M114 works on pronsole, but I can't figure out how to relate that to printcore.

Thanks for the help

CharlesFr, 2018-10-03:
Printcore could potentially be such a useful module but the lack of documentation and simple functionality renders it pretty much useless. I've resorted to re-writing it from scratch with the functions I need.

rockstorm101, 2018-11-25:
> I've resorted to re-writing it from scratch with the functions I need.

So I tried to implement such a feature which seems to be requested quite often. While trying to get my head around how Printcore works I ended up kind of rewriting it myself too. I know it probably means reinventing the wheel but I've been thinking a lot about this and maybe a re-implementation is required and could beneficial.

@CharlesFr, Would you please share your work?

I've been working not only on such re-implementation [[1]] but also on adding proper unit testing and optimizing it in terms of memory usage. So now would be the right time for discussing what the API should look like. I uploaded a very first draft of it [[2]]. What would you expect to find in Printcore's API? Any comments or suggestions would be more than welcomed.

[1]: https://github.com/rockstorm101/Printrun/blob/reimplementation/printrun/printcore.py
[2]: https://gist.github.com/rockstorm101/0527ca34c2533027484fc65f37593166

If this is too crazy and I shouldn't be wasting my time with this, please let me know too :P. 

(CC @kliment, @iXce, @hroncok)

kliment, 2018-11-25:
I really like where you're going with this, here's a few initial comments:

The core difficulty that people trying to use the existing code have is understanding that the communication is all asynchronous with timing driven by the receiving end. New commands get put on a queue, and responses trigger a callback.

This is also what makes implementing your proposed change difficult - responses can happen several commands after the initial command was sent, and there is nothing to identify which response was to which command - counting them is also impractical because of possible resends. The way it's currently implemented we use the contents of the response itself to tell what command it's supposed to be a response to. I don't know how to implement this better but I'd love for it to happen.

rockstorm101, 2018-11-26:
> The core difficulty that people trying to use the existing code have is understanding that the communication is all asynchronous with timing driven by the receiving end. New commands get put on a queue, and responses trigger a callback.

I completely agree, this is what seems to generate most of the confusion. Then this detail should either be hidden by the implementation or explained in docs. The latter I'll probably do anyway.

> This is also what makes implementing your proposed change difficult - responses can happen several commands after the initial command was sent, and there is nothing to identify which response was to [...]

I can't find where exactly did I read about it last time but from these posts [[1]][[2]] one can infer that what they call the "ping-pong" method is safe, at least for the firmwares Marling and Grbl. So one can simply send a command, wait for 'ok', send another command, wait for 'ok',... Again I'm unable to find the correct documentation now, I'm sorry, but this post [[3]] shows the firmware has it own command buffer. So, if I remember correctly from Marlin docs, the firmware will receive the command, put it in the buffer, and send the 'ok' without waiting. Only when this buffer is full, the firmware won't respond an 'ok' until it finishes processing the current command. So the new API could provide something like two functions:

- `send_nowait`: like the current `send_now` that sends the command ahead but does not return anything.
- `send_and_wait` (name open for improvement): that would send a command ahead of the queue, wait for a response, and return the response (it would be up to user to process this response and convert it into a array of coordinates or whatever).

Another advantage of the "ping-pong" method is that if a command is to be resent, just the last command has to be resent, and not a bunch of them from a few commands back. 

[1]: https://github.com/LaserWeb/deprecated-LaserWeb3/issues/2#issuecomment-223545963
[2]: https://github.com/LaserWeb/deprecated-LaserWeb3/issues/3#issuecomment-223384503
[3]: https://3dprinting.stackexchange.com/a/230.

I know the biggest drawback of this method is probably random pauses due to the host catching up with the CNC machine or vice versa. Sadly, I don't have my printer at hand to test this at the moment. But, as explained by the firmware developers in the previous posts, the "ping-pong" method should be good enough.

CharlesFr, 2018-11-29:
> The core difficulty that people trying to use the existing code have is understanding that the communication is all asynchronous with timing driven by the receiving end. New commands get put on a queue, and responses trigger a callback.

Should this not be made explicit in the documentation, in my mind this is the where the package falls short.

>@CharlesFr, Would you please share your work?

Unfortunately, it's for a commercial project - it's also been written specifically to the machine I'm working with so I'm not sure it'll be of much help. More than happy to help as I can though, I like what you've done.

> So, if I remember correctly from Marlin docs, the firmware will receive the command, put it in the buffer, and send the 'ok' without waiting. Only when this buffer is full, the firmware won't respond an 'ok' until it finishes processing the current command.

I still have some confusion on this point, the above is true if the command is a motion type such as `G0` and `G1`. However, commands such as `M114` seem to bypass the buffer and send a response right away. Can someone confirm this is the case? If so, is there a list of commands with this behavior?

kliment, 2018-11-29:
> I still have some confusion on this point, the above is true if the
> command is a motion type such as |G0| and |G1|. However, commands such
> as |M114| seem to bypass the buffer and send a response right away. Can
> someone confirm this is the case? If so, is there a list of commands
> with this behavior?

This is down to firmware implementation and is not universal - the way
Marlin works for example is by reporting where the machine will be when
it has executed all the commands currently in its buffer, not where the
machine is now. The reply comes immediately, but describes a state that
is in the future. I'm not really qualified to describe how every command
works in every implementation of the firmware, and it's not something we
can reasonably keep track of in printrun.

CharlesFr, 2018-11-29:
Thanks for the clarification.

amrithmmh, 2019-02-15:
@CharlesFr  i am also trying to get output after sending command say M114 but there is no proper documentation can you please share how you got the feedback from printer?

amrithmmh, 2019-02-15:
> Printcore could potentially be such a useful module but the lack of documentation and simple functionality renders it pretty much useless. I've resorted to re-writing it from scratch with the functions I need.

please share the python class that is relevant i have the same problem only need to send a simple command and recived feedback