kliment/Printrun #966: Printcore error trying to run code after the print

skwhite16, 2018-10-09:
I am using printcore to communicate with my printer via a python script.When I try to run additional commands after printing my gcode file I get the following error:
>>> ERROR:root:Print thread died due to the following error:
Traceback (most recent call last):
  File "C:\Users\Sydney\Desktop\PrintrunMaster\Printrun-master\printrun\printcore.py", line 594, in _print
    self._sendnext()
  File "C:\Users\Sydney\Desktop\PrintrunMaster\Printrun-master\printrun\printcore.py", line 643, in _sendnext
    gline = self.mainqueue.all_layers[layer][line]
IndexError: list index out of range

My python code looks like this:
 gcode = [i.strip() for i in open('gUnit.gcode')]
 gcode = gcoder.LightGCode(gcode)
 p.startprint(gcode)
 p.send("G90")

The printer will execute the  p.send("G90") line but the error means that I have to rerun the whole python script in order to continue using the printer. How do I avoid the error?

rockstorm101, 2018-11-24:
Hi @skwhite16, did you find an answer to your problem? Could please share your solution? If you could not fix it, I might be now in a position to help.

skwhite16, 2018-11-26:
The problem had something to do with indexing the commands to the printer between what was in the gcode file and the commands sent after through python. I think it was expecting a certain number of lines in the gcode file and then got messed up by the additional command. I just added a delay that was equal to the time it took to run my gcode file:

p.startprint(gcode)
sleep(11)
p.send("G90")

Probably not the greatest solution but solved the issue.

rockstorm101, 2018-11-26:
> The problem had something to do with indexing the commands to the printer between what was in the gcode file and the commands sent after through python. 

Could you please test what happens if you use `send_now` instead of `send`? So your code would look like this:
```
gcode = [i.strip() for i in open('gUnit.gcode')]
gcode = gcoder.LightGCode(gcode)
p.startprint(gcode)
p.send_now("G90")
```
Thanks in advance.