kliment/Printrun #967: no inicia

cahipe, 2018-10-11:
error inicio
Traceback (most recent call last):
  File "pronterface.py", line 66, in <module>
  File "printrun\pronterface.pyc", line 2296, in __init__
  File "printrun\pronterface.pyc", line 220, in __init__
  File "printrun\pronterface.pyc", line 279, in reload_ui
  File "printrun\gui\__init__.pyc", line 155, in createTabbedGui
  File "printrun\gui\controls.pyc", line 299, in add_extra_controls
  File "printrun\gui\controls.pyc", line 126, in add
  File "wx\_core.pyc", line 14086, in Add
wx._core.PyAssertionError: C++ assertion "!CheckForIntersection(item)" failed at ..\..\src\common\gbsizer.cpp(225) in wxGridBagSizer::Add(): An item is already at that position

rockstorm101, 2018-10-18:
Hi @cahipe, sorry for responding so late. Please have a look at [this answer][1] and let me know if it solves your problem.

[1]: https://github.com/kliment/Printrun/issues/879#issuecomment-359346976

cahipe, 2018-10-19:
Yes thanks
I managed to fix the error and it starts perfectly.
Thank you.

closed 2018-10-19 by rockstorm101