kliment/Printrun #968: Unable to input parameters into printrun

lsippell, 2018-10-20:
Can not input 
extrude  length or speed
print speed or flow %
printer width, depth or height
Z, Y or Z offset nor home position
Tried both Printrun-Win-Slic3r-03Feb2015 and Printrun-win-18Nov2017
Running Win10
Previously run on this computer with Win10. Stopped after not using it for several weeks
[image]

rockstorm101, 2018-10-20:
Hi @lsippell, a similar problem was reported before and [this solution][1] was supposed to work. Could you please try and let us know it if helps? Also, could please confirm whether the same behaviour is experienced using the [latest beta release][2]?

[1]: https://github.com/kliment/Printrun/issues/856#issuecomment-354091191
[2]: https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc5

lsippell, 2018-10-21:
The solution worked!
The beta release worked correctly without the changes.

closed 2018-10-21 by lsippell