kliment/Printrun #970: GCode Arcs not displayed correctly

hurzl, 2018-11-01:
In the 2D preview (at least) arcs sometimes go the wrong direction, showing huge nonsense arcs. The Marlin firmware prints them ok.

[image]

rockstorm101, 2018-11-02:
Hi I'm trying to reproduce this. Could you please share an example file with this kind of problem? Both the stl and the gcode?

hurzl, 2018-11-02:
I suspect the ccw (G3) arcs ... if you remove them the circles are gone ...

hurzl, 2018-11-27:
you may be getting errors with arcs anyway?
I have to apply following patch
``` 
------------------------------- printrun/gviz.py -------------------------------
index df3ea71..8825e3f 100644
@@ -320,6 +320,7 @@ class Gviz(wx.Panel):
         dc.SetBrush(wx.TRANSPARENT_BRUSH)
         for i in range(len(scaled_arcs)):
-            dc.SetPen(pens[i] if isinstance(pens, list) else pens)
+#            dc.SetPen(pens[i] if isinstance(pens, list) else pens)
+            dc.SetPen(pens[i] if hasattr(pens, "__getitem__") else pens)
             dc.DrawArc(*scaled_arcs[i])
 
     def repaint_everything(self):
```
to get rid of error messages like
```
TypeError: DC.SetPen(): argument 1 has unexpected type 'numpy.ndarray'
```