kliment/Printrun #972: Z lift breaks layer detection

rb1205, 2018-11-14:
When I add some Z lift to retractions in Simplify3D, Printrun sees every retraction as a layer change, which messes up the gcode view and makes hard to use features like the excluder. Attached is an exmple of Simplify3D gcode with 0.05mm Z lift on retraction:

```
G1 X200.024 Y168.550 E0.4003
G1 E-0.8000 F1800
G1 Z1.450 F200
G1 X211.300 Y188.651 F9000
G1 Z1.400 F200
G1 E0.8000 F1800
G1 X213.877 Y191.228 E0.0873 F2700
G1 X213.818 Y192.125 E0.0215
```

I think that they can be easily detected if you look for small Z changes with an XY non-printing movement in the middle.

rockstorm101, 2018-11-21:
Hi @rb1205. I think this is something to be addressed in [gcoder.py][1]. The file says:

`This file is copied from GCoder.`

But I can't find the "original" file or project. @kliment, where was this file taken from?

[1]: https://github.com/kliment/Printrun/blob/master/printrun/gcoder.py

kliment, 2018-11-25:
Hey, sorry I didn't reply earlier, I just got back from Shenzhen. I believe the original code is by Mikko Sivulainen ( @sivu ), who then merged it into Printrun at commit 577ed7c8fb8eaf4d40e059384be03df0e5a9e307 , and Printrun is now its official home. I have no knowledge of gcoder existing as an upstream project independently of printrun.

rockstorm101, 2018-11-26:
>  I believe the original code is by Mikko Sivulainen ( @sivu ), who then merged it into Printrun at commit 577ed7c , and Printrun is now its official home.

Alright, thanks for the info. Then is safe for us to modify this file instead of creating an issue in some other project.