kliment/Printrun #976: numpy is actually a required dependency

ejona86, 2019-01-07:
1.6.0 and later (I also tested 2.0.0rc5) have a required dependency on numpy. Running without numpy installed causes pronterface to hang before opening its window. It seems this was introduced in c178ad0cd.

The README.md (for 1.x and master) still implies that numpy is optional for 3D view support. It seems either the readme or gviz.py should be fixed. Any preference?

Log running 1.6.0:
```
$ pronterface.py 
08:57:29: Warning: Mismatch between the program and library build versions detected.
The library used 3.0 (wchar_t,compiler with C++ ABI 1013,wx containers,compatible with 2.8),
and wxPython used 3.0 (wchar_t,compiler with C++ ABI 1011,wx containers,compatible with 2.8).
/usr/lib/python2.7/site-packages/wx-3.0-gtk3/wx/_core.py:16629: UserWarning: wxPython/wxWidgets release number mismatch
  warnings.warn("wxPython/wxWidgets release number mismatch")
WARNING:root:Could not setup DBus for sleep inhibition: No module named dbus
WARNING:root:psutil unavailable, could not import power utils:No module named psutil
[ERROR] 3D view mode requested, but we failed to initialize it.
Falling back to 2D view, and here is the backtrace:
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/printrun/gui/viz.py", line 70, in __init__
    import printrun.gcview
  File "/usr/lib/python2.7/site-packages/printrun/gcview.py", line 22, in <module>
    from .gl.panel import wxGLPanel
  File "/usr/lib/python2.7/site-packages/printrun/gl/panel.py", line 21, in <module>
    import numpy
ImportError: No module named numpy

Traceback (most recent call last):
  File "/usr/bin/pronterface.py", line 66, in <module>
    app = PronterApp(False)
  File "/usr/lib/python2.7/site-packages/printrun/pronterface.py", line 2296, in __init__
    self.mainwindow = PronterWindow(self)
  File "/usr/lib/python2.7/site-packages/printrun/pronterface.py", line 220, in __init__
    self.reload_ui()
  File "/usr/lib/python2.7/site-packages/printrun/pronterface.py", line 282, in reload_ui
    self.settings.controlsmode == "Mini")
  File "/usr/lib/python2.7/site-packages/printrun/gui/__init__.py", line 256, in createGui
    viz_pane = VizPane(self, vizpanel)
  File "/usr/lib/python2.7/site-packages/printrun/gui/viz.py", line 79, in __init__
    from printrun import gviz
  File "/usr/lib/python2.7/site-packages/printrun/gviz.py", line 18, in <module>
    import numpy
ImportError: No module named numpy
```

Log running 2.0.0rc5:
```
$ pronterface.py 
WARNING:root:psutil unavailable, could not import power utils:No module named 'psutil'
3D view mode requested, but we failed to initialize it.
Falling back to 2D view, and here is the backtrace:
Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/printrun/gui/viz.py", line 70, in __init__
    import printrun.gcview
  File "/usr/lib/python3.7/site-packages/printrun/gcview.py", line 22, in <module>
    from .gl.panel import wxGLPanel
  File "/usr/lib/python3.7/site-packages/printrun/gl/panel.py", line 19, in <module>
    import numpy
ModuleNotFoundError: No module named 'numpy'

Traceback (most recent call last):
  File "/usr/bin/pronterface.py", line 62, in <module>
    app = PronterApp(False)
  File "/usr/lib/python3.7/site-packages/printrun/pronterface.py", line 2388, in __init__
    self.mainwindow = PronterWindow(self)
  File "/usr/lib/python3.7/site-packages/printrun/pronterface.py", line 218, in __init__
    self.reload_ui()
  File "/usr/lib/python3.7/site-packages/printrun/pronterface.py", line 295, in reload_ui
    self.settings.controlsmode == "Mini")
  File "/usr/lib/python3.7/site-packages/printrun/gui/__init__.py", line 255, in createGui
    viz_pane = VizPane(self, vizpanel)
  File "/usr/lib/python3.7/site-packages/printrun/gui/viz.py", line 79, in __init__
    from printrun import gviz
  File "/usr/lib/python3.7/site-packages/printrun/gviz.py", line 18, in <module>
    import numpy
ModuleNotFoundError: No module named 'numpy'
```

hroncok, 2019-01-07:
https://github.com/kliment/Printrun/blob/master/requirements.txt has numpy.

So I guess we should stop saying numpy is optional.

ejona86, 2019-01-07:
requirements.txt also has cairocffi, cairosvg, dbus-python and maybe others that are indeed optional. It is clear that README has drifted from requirements.txt; cairo/cffi deps are different, for example.

rockstorm101, 2019-01-31:
> So I guess we should stop saying numpy is optional.

It actually is optional in a way. If you only run pronsole, you shouldn't need numpy, it is only needed for the GUI and the plater. Should it still be clarified?