kliment/Printrun #978: run_script and run_gcode_script crash pronsole/pronterface

Kuro-Maii, 2019-01-17:
when running the build in macro to run an external script both pronterface and pronsole will crash with the following:
```
run_gcode_script ./fan_on.py
Caught an exception, exiting:
Traceback (most recent call last):
  File "/home/angel/projects/other/Printrun/pronsole.py", line 54, in <module>
    interp.cmdloop()
  File "/home/angel/projects/other/Printrun/printrun/pronsole.py", line 244, in cmdloop
    stop = self.onecmd(line)
  File "/usr/lib/python3.5/cmd.py", line 217, in onecmd
    return func(arg)
  File "/home/angel/projects/other/Printrun/printrun/pronsole.py", line 1735, in do_run_gcode_script
    p = run_command(l, {"$s": str(self.filename)}, stdout = subprocess.PIPE, universal_newlines = True)
TypeError: run_command() got an unexpected keyword argument 'universal_newlines'

```
this is on the most recent master as of this writing

hroncok, 2019-01-17:
BTW we got this filed in Fedora Bugzilla as well: https://bugzilla.redhat.com/show_bug.cgi?id=1654399

There is a proposed fix, but something is still wrong with that.

I am already investigating, but I don't have much time currently to give into Printrun unfortunately.

hroncok, 2019-01-17:
Actually the link si https://bugzilla.redhat.com/show_bug.cgi?id=1589949

closed 2022-01-13 by volconst