kliment/Printrun #981: apt-get error Ubuntu 18.04

brosskgm, 2019-01-31:
sudo apt-get install python-serial python-wxgtk2.8 python-pyglet python-numpy cython python-libxml2 python-gobject python-dbus python-psutil python-cairosvg libpython-dev git

Reading package lists... Done
Building dependency tree       
Reading state information... Done

Package python-wxgtk2.8 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'python-wxgtk2.8' has no installation candidate

brosskgm, 2019-01-31:
On a link for the install. Let me try and find it again for you and send you the link.

brosskgm, 2019-01-31:
Here you go.

https://github.com/kliment/Printrun/commit/3d495baba83dd7a4e4388b2ac4e2ca552c2143e0

brosskgm, 2019-01-31:
I used a different page after that error and got it installed and running. Just wanted to let you know about this one. There is another page with the same string on it. Still looking for it.

rockstorm101, 2019-01-31:
> I used a different page after that error and got it installed and running. Just wanted to let you know about this one.

I understand the problem is solved then.

For reference, please use the [updated installation documentation](https://github.com/kliment/Printrun#running-from-source)

> [3d495ba](https://github.com/kliment/Printrun/commit/3d495baba83dd7a4e4388b2ac4e2ca552c2143e0) actually removes those lines.

Yeah, I don't really understand this.

closed 2019-01-31 by rockstorm101