kliment/Printrun #983: Software Compatibility

MTUpower, 2019-02-05:
Dear Sir, I have been using Pronterface on my Mac for some time now, great works a treat.  But over the last two weeks I have been getting a message that the program may not work with future Updates of the Mac software. Have you got update planed to combat this?

rockstorm101, 2019-02-05:
Hi, could you please share what iOS version are you running, what version of Printrun, and maybe a screenshot of the message? Or if the message is shown in the console, could you please share it?

MTUpower, 2019-02-08:
Dear Sir

Please find below all the information you requested.

1.  macOS Mojave Version 10.14.2

2. Printrun  version 1.6

3. Message
[cid:C288D939-4616-4BE3-B041-CA7EDDD3E86D]

Hope this all helps

Duncan

rockstorm101, 2019-02-08:
Hi @MTUpower,

1. I cannot see the message, please upload it to GitHub instead of using the e-mail.
2. Could you please check whether using Printrun version 2.0.0rc5 [[1]] produces the same message?

[1]: https://github.com/kliment/Printrun/releases/download/printrun-2.0.0rc5/Printrun-mac-2.0.0rc5.dmg

MTUpower, 2019-02-08:
[image]

MTUpower, 2019-02-08:
Hello, quick question this new  Printrun version 2.0.0rc5 is this for Mac users?

rockstorm101, 2019-02-08:
> Hello, quick question this new Printrun version 2.0.0rc5 is this for Mac users?

I believe so. There should be a download link for iOS on the release page [[1]].

[1]: https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc5

Micho5233, 2019-09-09:
I can confirm that this does not work for the newest version of MacOS, Catalina. They dropped all support for 32 bit programs.