kliment/Printrun #985: RuntimeError: maximum recursion depth exceeded

eewc23, 2019-02-12:
I'm currently using pronterface to interact with a microfluidics pipetting robot instead of a 3d printer, just so you know. 

I have a number of custom buttons that move a motor attached to a syringe to aspirate and dispense liquid small volumes of liquid.  After making and testing a button to move the syringe plunger .075mm and thus dispense 50nl of fluid, none of my other buttons work.  Currently, if a press any button to move that motor at all in either direction and any distance, this is what spits out..
  File "<string>", line 3, in macro
  File "cmd.pyc", line 221, in onecmd
  File "printrun\pronsole.pyc", line 429, in <lambda>
  File "<string>", line 3, in macro
  File "cmd.pyc", line 221, in onecmd
  File "printrun\pronsole.pyc", line 429, in <lambda>
  File "<string>", line 3, in macro
  File "cmd.pyc", line 221, in onecmd
  File "printrun\pronsole.pyc", line 429, in <lambda>
  File "<string>", line 3, in macro
  File "cmd.pyc", line 221, in onecmd
  File "printrun\pronsole.pyc", line 429, in <lambda>
  File "<string>", line 3, in macro
  File "cmd.pyc", line 221, in onecmd
  File "printrun\pronsole.pyc", line 429, in <lambda>
  File "<string>", line 3, in macro
  File "cmd.pyc", line 221, in onecmd
  File "printrun\pronsole.pyc", line 429, in <lambda>
  File "<string>", line 3, in macro
  File "cmd.pyc", line 221, in onecmd
  File "printrun\pronsole.pyc", line 429, in <lambda>
  File "<string>", line 3, in macro
  File "cmd.pyc", line 221, in onecmd
  File "printrun\pronsole.pyc", line 429, in <lambda>
  File "<string>", line 3, in macro
  File "cmd.pyc", line 221, in onecmd
  File "printrun\pronsole.pyc", line 429, in <lambda>
  File "<string>", line 3, in macro
  File "cmd.pyc", line 221, in onecmd
  File "printrun\pronsole.pyc", line 429, in <lambda>
  File "<string>", line 2, in macro
RuntimeError: maximum recursion depth exceeded

At first I wondered if the motor was incapable of moving such a short distance, but then I remembered that I have a previously working button that moved it .01mm.  
Any ideas?

volconst, 2019-06-27:
The problem could be in the body of the macro, if the macro calls itself.
Can you post an example which reproduces the error? Which version are you using?
Try renaming the macro.

rockstorm101, 2020-05-09:
Closing due to a lack of feedback. Please feel free to reopen if needed.

closed 2020-05-09 by rockstorm101