kliment/Printrun #988: How can I pull the memory on the Arduinos RAM through printrun?

skwhite16, 2019-03-12:
My printer runs on a RAMPS/arduino mega, firmware is repetier. I am controlling it via python script using printrun/printcore. I want to be able to check the currently allocated memory on the arduino RAM through my python script. Any advice?

a2k-hanlon, 2021-02-03:
Hi @skwhite16, sorry your question didn't get answered sooner. Unfortunately I do not believe this is possible using Printrun. Printrun and your printer only speak g-code with each other, and there is no way to ask for raw contents of microcontroller RAM via g-code.

Reading a running microcontroller's memory is typically done with a hardware debugger. As far as I know, interfacing a hardware debugger with many Arduino boards is difficult because Arduinos are not designed with this in mind.

I am going to close this issue now because there's been no activity here for some time, but please feel free to re-open if you have further questions, or open a new issue.

closed 2021-02-03 by a2k-hanlon

volconst, 2021-02-03:
The question is probably just for the size of the memory.
Try https://marlinfw.org/docs/gcode/M100.html

a2k-hanlon, 2021-02-03:
Oh, thanks @volconst. In that case I misunderstood the question, and good suggestion.