kliment/Printrun #996: Inconsistent behavior using printcore between IDLE and command prompt

jlauden, 2019-04-05:
Apologies to all if this is a simple fix, I'm a little new to all of this.

I cloned the git repository into a folder in Documents and created a python script that I'd like to use to run printcore.  I installed according to instructions, including Microsoft Visual C++.  I'm using Python 3.7.1.  I started with the example script from the readme and got the COM port and baud rate set up, and set it to open a gcode file I generated in Cura. 

Here's the script:
`from printrun.printcore import printcore
from printrun import gcoder

gc_file = "LM_20mm_hollow_cube.gcode"

p=printcore('COM4',250000)
p.connect('COM4',250000)
gcode=[i.strip() for i in open(gc_file)] # or pass in your own array of gcode lines instead of reading from a file
gcode = gcoder.LightGCode(gcode)
p.startprint(gcode) # this will start a print`

When running my script in IDLE, no error messages appear, the printer does not respond in any way, and it returns to the >>> almost instantly.  However, when I use the send_now() command  after the script has run, the printer responds.  So it is connecting properly.

When using the command prompt (with or without entering the virtual environment), the cursor moves to the next line and hangs out there, like it is working on something, but nothing happens.  

Finally I tried setting a breakpoint on the last two lines of the script and using the debugger.  Executing the lines through the debugger, the printer started happily working it's way through the gcode without issue.  

Any ideas?  I am really grateful that this software is out there, just trying to tinker long enough to get it to work.  My end goal is to read the gcode and send a few lines at a time, read a couple sensors, modify upcoming gcode if needed, and send the next few lines.

jlauden, 2019-04-05:
Okay so after posting all that, importing time and adding sleep(3) a line above startprint() seems to solve the issue.  Still, might be  something that could or should be corrected?

kliment, 2019-04-05:
This is a problem with how idle handles threads. It shows up a lot in
gui work as well. We can't do anything about it sadly.

hrj, 2019-11-21:
@jlauden I ran into the same problem. Thanks for posting the `sleep` solution.

I optimised it a little bit:

```
 while not p.online:
   sleep(0.1)
```

This will poll the printer to see if the connection succeeded and wait for 100 milliseconds before polling again. This will allow quicker printing in case the printer does connect earlier than the hard-coded value of 3 seconds.

a2k-hanlon, 2021-02-02:
@kliment is there a reason this is still open if it's a "won't fix" issue?

kliment, 2021-02-02:
no reason, you can close it

closed 2021-02-02 by a2k-hanlon