kliment/Printrun #997: Elementary OS

Asada1, 2019-04-13:
how can i instal pronterface on elementary os? ther no pakage https://extras.wxpython.org/wxPython4/extras/linux/    for me

rockstorm101, 2019-04-13:
On April 13, 2019 12:32:22 PM GMT+01:00, Asada1 <notifications@github.com> wrote:
>how can i instal pronterface on elementary os? ther no pakage
>https://extras.wxpython.org/wxPython4/extras/linux/    for me

Hi, I don't know anything about Elementary OS but since it is derived from Ubuntu, that one might work. If you try it, please post here your findings for future reference :)

Asada1, 2019-04-14:
Ok, I have know that elementary is similar to ubuntu, and same times to debian too.
I have try install that for ubuntu 14;16;18 and all debian and terminal allways answer me in same error(i don’t remember it now)

rockstorm101, 2019-04-14:
Could you please confirm there is no package named `python3-wxgtk4.0` in Elementary OS's package repository?
```shell
sudo apt update
apt search python3-wxgtk4.0
```

Asada1, 2019-04-14:
Yes ther is. I’ll write if that start work

Asada1, 2019-04-14:
Error: Command '['/home/asada/Downloads/Printrun/Printrun/venv/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip']' returned non-zero exit status 1.

rockstorm101, 2020-04-18:
Hi @Asada1, I know the issue is quite old, but were you able to solve this problem? In case you did could you please share the outcome and close this issue?

rockstorm101, 2020-05-23:
Due to the lack of feedback I assume the given solution worked. Please feel free to reopen if needed.

closed 2020-05-23 by rockstorm101