kliment/Printrun #998: Call Pronterface in MATLAB to initiate 3D printer movement in a system?

OHaleNo, 2019-04-18:
Hi,
My goal is to use Printrun to talk to my 3D printer via USB for robotic movements (we want to use the 3D printer arm for magnetic field mapping by attaching a magnetometer to it).  While I can run my own GCode with Printrun by loading a text file I've written into it, I would like to know how I can call Printrun in a MATLAB script that is running many other devices in our system.  Essentially, I am hoping to find a way to call a Python script that is a part of Printrun in MATLAB and input desired GCode instructions in MATLAB that will then initiate the Python to move the 3D printer via USB.  Is this possible?  And if so, what part of Printrun would be most useful for this (Pronterface, Pronsole, or Printcore)?
Thanks!

benjaminforest, 2019-04-18:
I think you'll need pronsole for this, since it is the command line interface for printrun.
Maybe calling it as an external program with matlab (https://fr.mathworks.com/help/matlab/matlab_external/run-external-commands-scripts-and-programs.html) and finding a way to pass gcode commands to it while it is running could do the job ?

benjaminforest, 2019-04-18:
Printcore seems more dedicated to creating your custom printer controller app

OHaleNo, 2019-04-18:
Hi benjaminforest,
I agree- I have been trying to get Pronsole up and running but I just posted that I am having some issues with getting it to launch.  Is Pronsole.py supposed to launch a command window of it's own?  Or is it called through Command Prompt or IDLE itself?
And thank you for the feedback- I am hoping this is possible and am glad to hear someone else has the same thoughts.
Best!

OHaleNo, 2019-04-18:
> Printcore seems more dedicated to creating your custom printer controller app

Hi benjaminforest,
I agree- I have been trying to get Pronsole up and running but I just posted that I am having some issues with getting it to launch. Is Pronsole.py supposed to launch a command window of it's own? Or is it called through Command Prompt or IDLE itself?
And thank you for the feedback- I am hoping this is possible and am glad to hear someone else has the same thoughts.
Best!

kliment, 2019-04-19:
You should be able to use this https://www.mathworks.com/help/matlab/matlab_external/call-python-from-matlab.html to create a printer object using printcore and then call its connect and send_now functions. I am not sure how to handle the callbacks in matlab so you won't have feedback on what it's doing unless you can figure that out.

I don't know how to tell matlab where to look for python modules but here is what the python code involved would look like:

`p=printcore.printcore("COM4",250000) #or whatever your printer is attached to`
`while not p.online: sleep(1) #p.online will become true once connected to printer`
`p.send_now("G1 X10 Y20") #or whatever`

OHaleNo, 2019-04-24:
Thanks so much.  I appreciate your feedback.  The idea is that I don't want to have to load  a 3D object and just send Gcode coordinate instructions- would that still be doable using Pronsole this way?

OHaleNo, 2019-04-24:
I'm still having much difficulty getting Pronsole to launch which is strange.