kliment/Printrun #999: Pronsole command prompt isn't launching?

OHaleNo, 2019-04-18:
Hi,
I have done all of the set up required for Pronsole (Python3.6, pyserial, pyreadline) but every time I try to launch Pronsole, the command window quickly pops up and disappears.  I can hardly read it but I believe it says something about not enough memory but I know that isn't true for my computer.  Has this happened to anyone before?  Am I missing an install/setup step?
Thanks!

rockstorm101, 2019-04-23:
Hi @OHaleNo, could you please tell us what happens if you invoke pronsole directly from the terminal? I mean typing `pronsole` at the terminal instead of double clicking the executable.

OHaleNo, 2019-04-24:
Hi.  Thank you for your feedback.  Upon re-downloading and re-setting up, I tried it in command prompt and got this:

C:\Users\WMPTeam\Desktop\Printrun-win-18Nov2017>pronsole
13:23:43: Debug: src/helpers.cpp(140): 'CreateActCtx' failed with error 0x00000008 (not enough memory resources are available to process this command.).

I tried Pronterface and that works fine.

OHaleNo, 2019-04-25:
Ok so I got Pronsole up and running- I type help and get the command options.  However, it cannot find my printer even with the correct COM port and Baud rate.  I get "Printer is not online.  Unable to move" when I send a move command.  Any advice?  Am I missing something?

OHaleNo, 2019-04-25:
All set- different Baud rate for some reason.

closed 2019-04-25 by OHaleNo

rockstorm101, 2019-04-26:
On April 25, 2019 11:52:16 PM GMT+01:00, OHaleNo <notifications@github.com> wrote:
>Ok so I got Pronsole up and running- I type help and get the command
>options.  However, it cannot find my printer even with the correct COM
>port and Baud rate.  I get "Printer is not online.  Unable to move"
>when I send a move command.  Any advice?  Am I missing something?

Could you please share what you did to fix it?

Thanks

OHaleNo, 2019-04-26:
Hi,
I googled the issue and had to do some things in Regedit.  Here is the video I watched.
https://www.youtube.com/watch?v=gV3Cqo-8v7Q
thanks