kliment/Printrun #1000: Configuration elements not visible in qt5

avolkov, 2019-04-21:
[image]

I'm running KDE 5 in debian testing/unstable and current theming makes it nearly impossible to update values of extrusion length, speed print speed and print flow. (See screenshot above).

I've tried to use --style option to revert to gtk theming, but that didn't work.

Is it possible to adjust element size/dimension so they appear correctly in KDE or an option to disable theming althogether.

I use printrun version 1.6.0

rockstorm101, 2019-04-23:
Hi @avolkov, could you please try version 2.X and let us know if it looks the same?

avolkov, 2019-04-23:
@rockstorm101 I'm unable to test 2.X version as I can't install dependencies for the wxWidgets on my system (Debian testing).

I keep get the following error when trying instal wxwidgets dependencies in pyenv virtualev (python version 3.7.2).

Running `pip install -r requirements.txt` produces the following.

```
    Checking for /tmp/pip-install-13hge5o4/wxPython/bin/waf-2.0.8...
    "/home/alex/.pyenv/versions/3.7.2/envs/printrun/bin/python3.7" /tmp/pip-install-13hge5o4/wxPython/bin/waf-2.0.8 --wx_config=/tmp/pip-install-13hge5o4/wxPython/build/wxbld/gtk3/wx-config --gtk3 --python="/home/alex/.pyenv/versions/3.7.2/envs/printrun/bin/python3.7" --out=build/waf/3.7/gtk3 configure build
    Setting top to                           : /tmp/pip-install-13hge5o4/wxPython
    Setting out to                           : /tmp/pip-install-13hge5o4/wxPython/build/waf/3.7/gtk3
    Checking for 'gcc' (C compiler)          : /usr/bin/gcc
    Checking for 'g++' (C++ compiler)        : /usr/bin/g++
    Checking for program 'python'            : /home/alex/.pyenv/versions/3.7.2/envs/printrun/bin/python3.7
    Checking for python version >= 2.7.0     : 3.7.2
    python-config                            : /home/alex/.pyenv/shims/python3.7-config
    Asking python-config for pyembed '--cflags --libs --ldflags' flags : yes
    Testing pyembed configuration                                      : yes
    Asking python-config for pyext '--cflags --libs --ldflags' flags   : yes
    Testing pyext configuration                                        : Could not build python extensions
    The configuration failed
    (complete log in /tmp/pip-install-13hge5o4/wxPython/build/waf/3.7/gtk3/config.log)
    Command '"/home/alex/.pyenv/versions/3.7.2/envs/printrun/bin/python3.7" /tmp/pip-install-13hge5o4/wxPython/bin/waf-2.0.8 --wx_config=/tmp/pip-install-13hge5o4/wxPython/build/wxbld/gtk3/wx-config --gtk3 --python="/home/alex/.pyenv/versions/3.7.2/envs/printrun/bin/python3.7" --out=build/waf/3.7/gtk3 configure build ' failed with exit code 1.
    Finished command: build_py (0m1.349s)
    Finished command: build (3m14.582s)
    Command '"/home/alex/.pyenv/versions/3.7.2/envs/printrun/bin/python3.7" -u build.py build' failed with exit code 1.
    
    ----------------------------------------
Command "/home/alex/.pyenv/versions/3.7.2/envs/printrun/bin/python3.7 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-13hge5o4/wxPython/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-w8k7rdz3/install-record.txt --single-version-externally-managed --compile --install-headers /home/alex/.pyenv/versions/3.7.2/envs/printrun/include/site/python3.7/wxPython" failed with error code 1 in /tmp/pip-install-13hge5o4/wxPython/
```

I suspect this might be the cause -- https://github.com/wxWidgets/Phoenix/issues/904

Is there a python wheel I can run on my system?

rockstorm101, 2020-04-18:
Hi @avolkov, I know the issue is quite old, but were you able to solve this problem? In case you are still using Printrun, does this keep happening with the latest release?