kliment/Printrun #1003: Fix the windows instructions for building

janissimsons, 2019-05-15:
Instructions for building source code for a windows machine are wrong. I set up python 3 from scratch just as instructed, installed virtualenv package, set up a virtual env inside the source repository. After activating and downloading all packages inside requirements.txt, the pronterface.py crashes on startup. 
First it crashed because it didn't have wxPython >= 4 installed, even though after checking it did have this version, but fine, I installed the package globally.
Then it complained about missing AppDirs folder... at that point I gave up.
Please fix your instructions for windows machines.

volconst, 2019-06-23:
I just followed buildinstructions.txt for windows and ran pronterface.
Maybe you have tried to run pronterface.py without activating the environment. This way you have installed the requirements in the environment, but executed later outside environment, hence you had to install wxpython again globally.

volconst, 2019-06-26:
I saw today that if I run just 'pronterface.py' from the command line I get wx missing error, even with activated environment, but when I run it with 'python pronterface.py' as per buildinstructions.txt, it works fine.

AakaFosfor, 2019-11-09:
I have no problem with this - I've built, run and packed the pronterface from sources on Win7 32bit according to the buildinstructions.txt just now without any problem.

kliment, 2020-03-31:
This seems to be resolved - please reopen if it's still a problem

closed 2020-03-31 by kliment