kliment/Printrun #1006: RC5 - "Failed to execute script pronterface" when launching

adiastra, 2019-05-30:
When launching pronterface-windows-2.0.0rc5.exe I get the error "Failed to execute script pronterface" and the program does not launch. I have trued running as admin and compatibility modes, same issue.

jackw01, 2019-06-09:
Same issue here on Windows 10 19H1. Was not happening a few weeks ago, and I haven't changed any configuration since then.

Korny-like, 2019-06-20:
I just switched to Insider in windows and have the same issue. It worked before.. So its windows update.. :-/

volconst, 2019-06-20:
What if you try to rebuild pronterface as described here https://github.com/kliment/Printrun/blob/master/buildinstructions.txt#L31

yufanyufan, 2020-07-13:
It works after rebuilding from code.

thbl, 2021-03-03:
+1 have same isue on latest release

thbl, 2021-03-03:
dosent work after i rebuild :(

virenchocha, 2021-10-14:
not everyone are pro and not every one can build...those instruction for building is goobly-gook for noob like me.

Tyrell63, 2022-01-04:
I have donweloaded the pronterface-windows-x64_3.8-2.0.0rc8 (also the x86 version). Tried both (also as administrator).....both with the same error. "failed to execute script pronterface"
OS is Windows 11

DivingDuck, 2022-01-04:
That's correct. The Windows version's are outdated due to wxPython 4.1.x incompatibilities with actual patched Windows10 and 11. 
You can check one of my latest interim compilations here: https://github.com/DivingDuck/Printrun/actions/runs/1596952500
The should not show this error (fingers crossed)

Please leaf a comment here if they work for you and add your OS information, like:

Windows Edition: Windows 10 Pro
Version: 21H2
Build: 19044.1415
OS Language: German (de_DE)

Tyrell63, 2022-01-05:
Hi

Sorry for the late reply,
But nw I was able to test it out.
I have downloaded the x64 3.9 version
And its running without any problems on my Windows 11 Enterprise, Version 10.0.22000 Build 22000 (English)

Thank you

Best,
Lutz

DivingDuck, 2022-01-08:
Thanks for your feedback.

Heinz-Loepmeier, 2022-05-20:
Had the same issue with the .exe from github releases.
Downloading latest from https://kliment.kapsi.fi/printrun/ fixed it.

DavidOx21, 2022-10-18:
The link to version 3.9 for windows 11 does not work and I cant find it any other way. Could someone send me a new link?

DivingDuck, 2022-10-18:
You can find my my latest interim test builds here: https://github.com/DivingDuck/Printrun/wiki/My-latest-development-version-build-with-github-workflow. They was build with wxPython 4.2 and Python 3.7, 3.8, 3.9 and 3.10.

closed 2023-04-03 by DivingDuck