kliment/Printrun #1007: Pronterface showing traceback error

oldegoraptor, 2019-06-06:
I ran into this issue a few times on multiple multi day prints.  So far I've only had this happen with this object.  The error wasn't generated until I updated my firmware on my Prusa I3 MK2.  This error showed up when slicing with Cura 3.0.4, and PrusaSlicer 2.0.  I was able to get past this issue by spamming pause/resume until the printer continued by itself.  I am unable to provide the STL and Gcode due to size, but would be more than happy to email them to support bug fixes.

Error message is as follows:

[ERROR] Print thread died due to the following error:
Traceback (most recent call last):
  File "printrun\printcore.pyc", line 534, in _print
  File "printrun\printcore.pyc", line 619, in _sendnext
  File "printrun\printcore.pyc", line 644, in _send
MemoryError

volconst, 2019-06-10:
This is a memory leak - dict sentlines is never freed. I believe this can be fixed with a macro even for running instances.

volconst, 2019-06-10:
Try this macro:
!print('Before free', len(self.p.sentlines))
!_max = 0
!_min = float('inf')
!for k in self.p.sentlines.iterkeys():
!    _max = max(_max,k)
!    _min = min(_min,k)
!print('min max', _min, _max)
!for k in xrange(_max-1024):
!    self.p.sentlines.pop(k, None)
!print('After free', len(self.p.sentlines))

It should be run when the MemoryError is about to happen or ealier, even 100 layers in the slicer if supported

CostanzoPablo, 2020-04-12:
Already fixed. You must put: (i tried on windows, and get working)

git clone
py -m venv venv
venv\Scripts\activate
python -m pip install -r requirements.txt
python pronterface.py

rockstorm101, 2020-04-18:
Hi @oldegoraptor, I know the issue is quite old, but were you able to solve this problem? Did you try the [given solution][1]?

[1]: https://github.com/kliment/Printrun/issues/1007#issuecomment-500525377

oldegoraptor, 2020-04-19:
I haven’t been able to mess with it for the last couple months due to a move from the EU.  But my stuff arrives here in the next week so I’ll be able to see if the fix mentioned before takes care of that issue.

rockstorm101, 2020-05-09:
Hi @oldegoraptor, I hope your moving went alright, do you have any updates on this?

davija, 2020-08-13:
When I got this error and printing stopped, I was able to hit pause, define and run the macro and then hit resume and it worked.  Thanks for the macro @volconst!

volconst, 2020-08-14:
Thanks for the feedback, @davija !
How large was the file in MB and lines?

davija, 2020-08-14:
Here is the file

[modified-fan-base3.zip](https://github.com/kliment/Printrun/files/5075930/modified-fan-base3.zip)

volconst, 2020-08-14:
2.1M lines, 67MB, thanks.