kliment/Printrun #1010: question about console debug communication log

sk8nfool, 2019-06-20:
Is the format of the log/console documented anywhere?

Digging a bit I found the following in printcore.py:

`
N<line_number> <command>*<chksum>
`

where
* `<line_number>` is line number (this is not the line number in the GCode file - not sure how lines are counted yet)
* `<command>` is GCode command being logged
* `<chksum>` is checksum of command (if I understand python its the logical sum (exor) rather than the arithmetic sum)

volconst, 2019-06-20:
The format of log is probably freestyle, not meant to be parsed.
The checksum is part of the communication protocol between the host and printer. it is parsed but the source code is the ultimate reference.
See http://marlinfw.org/docs/gcode/M110.html https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Marlin_main.cpp#L1084
https://reprap.org/wiki/Talk:G-code

a2k-hanlon, 2021-02-02:
Closing due to lack of activity.

closed 2021-02-02 by a2k-hanlon