kliment/Printrun #1012: Pronterface wont open on raspbery pi 3 b+

johanndavel, 2019-08-06:
So i installed from "sudo apt install printrun"
But when i try to run i get an error window saying "Invalid desktop entry file: '/home/pi/Printrun/pronterface.desktop'"
before that i installed from "git clone https://github.com/kliment/Printrun.git" and then it straight up just ignored me executing the py file. My question is how do i get to the point of pronterface launching on a pi3b+ ?

johanndavel, 2019-08-06:
Update. here is what i get when i execute it in terminal
`pi@raspberrypi:~ $ cd ~/Printrun
pi@raspberrypi:~/Printrun $ ./pronterface.py
wxPython >= 4 is not installed. This program requires wxPython >=4 to run.
Traceback (most recent call last):
  File "./pronterface.py", line 23, in <module>
    import wx  # NOQA
ModuleNotFoundError: No module named 'wx'
`

johanndavel, 2019-08-06:
pip install -U wxPython
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting wxPython
  Downloading https://files.pythonhosted.org/packages/9a/a1/9c081e04798eb134b63def3db121a6e4436e1d84e76692503deef8e75423/wxPython-4.0.6.tar.gz (68.8MB)
    99% |████████████████████████████████| 68.8MB 3.7MB/s eta 0:00:01Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/_internal/cli/base_command.py", line 143, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/_internal/commands/install.py", line 338, in run
    resolver.resolve(requirement_set)
  File "/usr/lib/python2.7/dist-packages/pip/_internal/resolve.py", line 102, in resolve
    self._resolve_one(requirement_set, req)
  File "/usr/lib/python2.7/dist-packages/pip/_internal/resolve.py", line 256, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "/usr/lib/python2.7/dist-packages/pip/_internal/resolve.py", line 209, in _get_abstract_dist_for
    self.require_hashes
  File "/usr/lib/python2.7/dist-packages/pip/_internal/operations/prepare.py", line 283, in prepare_linked_requirement
    progress_bar=self.progress_bar
  File "/usr/lib/python2.7/dist-packages/pip/_internal/download.py", line 836, in unpack_url
    progress_bar=progress_bar
  File "/usr/lib/python2.7/dist-packages/pip/_internal/download.py", line 673, in unpack_http_url
    progress_bar)
  File "/usr/lib/python2.7/dist-packages/pip/_internal/download.py", line 897, in _download_http_url
    _download_url(resp, link, content_file, hashes, progress_bar)
  File "/usr/lib/python2.7/dist-packages/pip/_internal/download.py", line 617, in _download_url
    hashes.check_against_chunks(downloaded_chunks)
  File "/usr/lib/python2.7/dist-packages/pip/_internal/utils/hashes.py", line 48, in check_against_chunks
    for chunk in chunks:
  File "/usr/lib/python2.7/dist-packages/pip/_internal/download.py", line 585, in written_chunks
    for chunk in chunks:
  File "/usr/lib/python2.7/dist-packages/pip/_internal/utils/ui.py", line 159, in iter
    for x in it:
  File "/usr/lib/python2.7/dist-packages/pip/_internal/download.py", line 574, in resp_read
    decode_content=False):
  File "/usr/share/python-wheels/urllib3-1.24.1-py2.py3-none-any.whl/urllib3/response.py", line 494, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "/usr/share/python-wheels/urllib3-1.24.1-py2.py3-none-any.whl/urllib3/response.py", line 442, in read
    data = self._fp.read(amt)
  File "/usr/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/filewrapper.py", line 63, in read
    self._close()
  File "/usr/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/filewrapper.py", line 50, in _close
    self.__callback(self.__buf.getvalue())
  File "/usr/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/controller.py", line 275, in cache_response
    self.serializer.dumps(request, response, body=body),
  File "/usr/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/serialize.py", line 55, in dumps
    "body": _b64_encode_bytes(body),
  File "/usr/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/serialize.py", line 12, in _b64_encode_bytes
    return base64.b64encode(b).decode("ascii")

rockstorm101, 2019-08-06:
Hi,

I assume you are running Raspbian, but I need to know what version. Could you please tell us the output of the command 'lsb_release -a'?

Cheers

johanndavel, 2019-08-06:
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

rockstorm101, 2019-08-06:
Great, installing package 'python3-wxgtk4.0' should solve one of the errors when running the git-cloned repository.
-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

johanndavel, 2019-08-06:
Sorry im a bit of a noob, i cant for the life of me find the package.

johanndavel, 2019-08-06:
sudo apt-get install python3-wxgtk4.0 
worked a charm thank you

rockstorm101, 2019-08-06:
> Sorry im a bit of a noob, i cant for the life of me find the package.

My bad, sorry. I should've been more specific.

> sudo apt-get install python3-wxgtk4.0
> worked a charm thank you

If Printrun is now running smooth and fine, could you please close the issue?

rockstorm101, 2020-04-18:
I assume the provided solution worked. Please feel free to reopen if needed.

closed 2020-04-18 by rockstorm101