kliment/Printrun #1013: Pronterface not starting

goldjunge91, 2019-08-25:
Hi 
On Win10 (1903 Build 18362.295)  Pronterface will not start.
[pronterface.exe.log](https://github.com/kliment/Printrun/files/3538041/pronterface.exe.log)

volconst, 2019-08-25:
Try to change some settings in pronterface.exe -> Right click -> Properties -> Compatibility

goldjunge91, 2019-08-26:
> Try to change some settings in pronterface.exe -> Right click -> Properties -> Compatibility

hello I tested it but it did not work
Win Vista 
Win 7  tested

volconst, 2019-08-26:
Try editing the printrunconf.ini as described in https://github.com/kliment/Printrun/issues/879
Are you using the latest release of pronterface?

rockstorm101, 2020-04-18:
I assume the [provided solution][1] worked. Please feel free to reopen if needed.

[1]: https://github.com/kliment/Printrun/issues/1013#issuecomment-525011963

closed 2020-04-18 by rockstorm101