kliment/Printrun #1016: Mac OS X Catalina (64 bit only)

KC5CQW, 2019-08-30:
Please make a 64 bit OS X release for compatibility with the latest MacOS. They have dropped all 32 bit support with this release.

Micho5233, 2019-09-09:
I'm in the same boat. My primary machine is an iMac, and in Catalina the is no workaround that can make Printrun work.

kliment, 2019-09-24:
Does this affect the https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc5 release as well?

Praktiker, 2019-10-05:
My Mac shows that printrun-2.0.0rc5 is 64bit. But I couldn't test it (not a beta tester).

Micho5233, 2019-10-05:
I was finally able to test RC5 on Catalina. Everything appears to work in this release! Thank you!

leonelhs, 2019-10-23:
Its very easy to build it from the sources just following the buildinstructions.txt 
Here is my build link: [Printrun-Mac-22Oct2019.zip](https://drive.google.com/open?id=1EM29GsKn4BgIRkGDUfsydpVr2sDctWuU)

lkvapil, 2020-01-02:
Thank you for your build, you saved me time 👍

3DMakerArgie, 2020-01-25:
Tahank you Kliment! You ended with more than 3 hours of researching how the H... compile pronterface in my MAC

numbakrrunch, 2020-05-02:
> Does this affect the https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc5 release as well?

No, I was having the same issue with the downloads here: http://kliment.kapsi.fi/printrun/ but the binary I found at this url works for me in OSX Catalina: https://github.com/kliment/Printrun/releases/download/printrun-2.0.0rc5/Printrun-mac-2.0.0rc5.dmg

MIchaeldm3, 2020-05-14:
THANK YOU NUMBA !!!

adolfoinigo, 2020-06-21:
it works for me thanks

LazaroFilm, 2020-07-10:
> Its very easy to build it from the sources just following the buildinstructions.txt
> Here is my build link: [Printrun-Mac-22Oct2019.zip](https://drive.google.com/open?id=1EM29GsKn4BgIRkGDUfsydpVr2sDctWuU)

Thank you! This worked for me.  This is a huge help. I wish Pronterface.com was properly updated with a 64 bit build.

M4R1OV2, 2020-08-05:
> Its very easy to build it from the sources just following the buildinstructions.txt
> Here is my build link: [Printrun-Mac-22Oct2019.zip](https://drive.google.com/open?id=1EM29GsKn4BgIRkGDUfsydpVr2sDctWuU)

listen i just created an account only to thank you, you just saved me

cdburgess, 2021-01-28:
I have confirmed this release also works on BigSur with Prusa 3.9.3 Firmware. Thanks!

rockstorm101, 2021-01-28:
> I have confirmed this release also works on BigSur with Prusa 3.9.3 Firmware. Thanks!

Hi @cdburgess, thank you for commenting. Could you please clarify what release is working for you? Is it the one provided [in the comment above][1] or is it the [latest 2.0.0rc7 release][2]? If is not the second one, would you mind giving it a try? (Might need to make use of a [workaround][3] to allow execution permission) 

[1]: https://github.com/kliment/Printrun/issues/1016#issuecomment-545267518
[2]: https://github.com/kliment/Printrun/releases/download/printrun-2.0.0rc7/Printrun-2.0.0rc7-macos.zip
[3]: https://github.com/kliment/Printrun/issues/1129#issuecomment-729151715

cdburgess, 2021-01-30:
I updated the comment. But I will download RC7 and try that too.

a2k-hanlon, 2021-02-02:
@cdburgess If it is not too much trouble, would you mind trying RC8 instead? It includes a number of significant changes relevant to the macOS build.

cdburgess, 2021-02-03:
Sure!  I will download it and try it out.

cdburgess, 2021-02-04:
RC8 is working just fine so far. No issues. Is there anything, in particular, you wanted me to test? I am able to connect to my Prusa and print files directly from my Mac. No issues at all. I can heat the bed too. I don't find any mechanism that will stop a print (mid print) that will raise the extruder and push the bed forward like it does when I stop a Prusa using the firmware. Am I missing something?

a2k-hanlon, 2021-02-04:
Thanks @cdburgess, I'm relieved to hear the the app runs smoothly on macOS 11! One thing I would like to know is whether or not Pronterface is able properly prevent the computer from entering sleep while printing, which prevents stalls.

There is a warning related to the macOS framework involved in this that comes up in the process of packaging the app, and I'm not sure if it actually indicates a problem not.

If you try to run a print for a duration longer than the time it would take for your computer to sleep automatically, and you don't otherwise keep your computer awake, does the print stall or does it complete without issues? If you have a chance to try this, please report back with a new issue if you run into problems. Thank you! 

I think that sequence of moves you mentioned when a print is stopped mid-print is implemented exclusively in the printer's firmware.

@kliment, @volconst anything else you would like to see tested on macOS?

3DMakerArgie, 2021-02-05:
I use a Mac to print with Pronterface/Printrun. To avoid sleeping, I leave
it connected to AC Power...and it doesn't sleep.

Este mensaje se ha enviado desde una casilla de GMAIL. Las casillas de
correo electrónico de GMAIL estan sujetas a violaciones de la privacidad y
monitoreo de la NSA o proveedores de servicios de Google. Téngalo en cuenta
si va a enviar información confidencial por este medio

volconst, 2021-02-05:
@cdburgess , @a2k-hanlon 
To execute custom gcode after pause
put in your pronsolerc file this line:
```
!self.pauseScript = 'on_pause.gcode'
```
create a file named on_pause.gcode in the current directory for pronterface and put your gcode e.g.:
```
G91;
G0 Z5;
G0 X5 Y5;
; or
; G28 X;
```
Currently there is a bug, which strips the last character of the lines, so the ';' is needed to work around it.
If the binary distribution is used absolute filename must be specified.

011JHK, 2021-02-11:
Hi! Thank you for all this useful downloads and information!
I have updated to Big Sur on my MacBook Pro.
I can't seem to get my 3D printer to connect to the Port on Pronterface...
it gives me 2 options: /dev/tty.usbmodem14201  or. /dev/cu.usbmodem14201 but neither would connect.
Am I missing something?

cdburgess, 2021-02-11:
> Hi! Thank you for all this useful downloads and information!
> I have updated to Big Sur on my MacBook Pro.
> I can't seem to get my 3D printer to connect to the Port on Pronterface...
> it gives me 2 [...]

What printer are you using and how is it connected to your Mac?  I am running Big Sur on my MacMini connecting via USB to my Prusa MK2.5s and it connects just fine. What baud rate are you using? I'm set at 115200.

randolphfrusso, 2021-02-17:
> What printer are you using and how is it connected to your Mac? I am running Big Sur on my MacMini connecting via USB to my Prusa MK2.5s and it connects just fine. What baud rate are you using? I'm [...]

I am also having this same issue. I am running a Creality CR-10 V2. With me, I have the printer connected, Pronterface open, and i have tried both the cu. and tty. ports as shown above. When i click connect, I know it does something on the printer as it restarts the screen. but then in pronterface it keeps saying "Got rubbish reply from..." "Maybe a bad baud rate?" I have tried each baud rate option and none of them work. 

Any ideas?

a2k-hanlon, 2021-02-19:
Hi @randolphfrusso, could you please clarify which OS version you are using, and which version of Printrun you are using? Is this a problem that has come up after you updated something? Have you tried different USB cables, or different USB ports on your computer? You may want to open a new issue since this one is getting rather crowded.

a2k-hanlon, 2021-03-07:
@cdburgess I have now been able to confirm on my own that the sleep prevention functionality I mentioned earlier on is working for the packaged app, so no need to worry about checking that. Thanks again for your help.

Amakin21, 2021-03-17:
Hey friends, total newbie here so sorry if this is redundant. I've never explored GitHub but I am trying to run pronterface on my Mac. I downloaded the link y'all mentioned and it pulled and opened pronter, but for some reason when I type in the codes, I'm not seeing the responses open up. I know the commands are working though bc I just ran a PID autotune via the M303 E0 S200 C8, and my printer ran through the cycle. but the values never populated in the dialogue box and it never told me PID complete. Do I need to download other files? Or am I doing this wrong? Appreciate any and all help

a2k-hanlon, 2021-03-17:
Hi @Amakin21, please open a new issue with your questions and I will do my best to help there.

When you do this, please try to share as much information as you can that is relevant to the problems you are encountering. For example, what model of printer are you using, what firmware does it run, how is it connected to your computer, and which version of a) Printrun and b) macOS are you are using?

a2k-hanlon, 2021-03-17:
I am closing this issue now since the original problem, the absence of a 64-bit binary for macOS, has been resolved. See https://github.com/kliment/Printrun/releases. Furthermore, all outstanding comments (other than @Amakin21's) have either been resolved, or have not been followed up on in at least a month as of writing.

closed 2021-03-17 by a2k-hanlon

DeannaTheBaker, 2021-04-07:
> What printer are you using and how is it connected to your Mac? I am running Big Sur on my MacMini connecting via USB to my Prusa MK2.5s and it connects just fine. What baud rate are you using? I'm [...]

I'm using Big Sur with an Ender 3v2, Micro Swiss Direct Drive, and BLTouch. I'm brand new to 3D printing, so not really understanding all of this. When I plug my usb in to my MacBook Pro, I'm not seeing any "usbmodem……" to  connect to. My baud rate is also set to the default 115200. I've seen a video that states I need a printer driver? Other videos don't mention that. I need any help you can give. Thank you.

volconst, 2021-04-07:
@DeannaTheBaker  You need a driver. It does just that - creates a device file through which you connect to the printer. The driver must be for the serial chip of your printer control board and for Big Sur. Inspect visually the control board for a version and chip and search the internet for the required driver. The name of device file can vary - compare the /dev/... files before and after driver installation.