kliment/Printrun #1020: Pronterface with MAC OS 10.14.6

NAP1947, 2019-09-29:
I am trying to install Pronterface on my Mac running OS 10.14.6.
I installs and opens and I can connect to my printer.
However, I cannot access any Mac Menu items when Pronterface is open.
Placing the cursor in any menu item and clicking nothing happens, cannot even access the Pronterface menu items.
Must be something simple I hope.

a2k-hanlon, 2020-05-04:
@NAP1947 I remember having this issue. The menu items would not appear when I opened Pronterface. It doesn't happen if you run Pronterface from the source code. What I found you can do to get to the menu items is "hide" and then "show" the Pronterface application using the dock icon right click menu. This is just a workaround though, it doesn't solve the underlying problem.

a2k-hanlon, 2021-02-02:
Hi again @NAP1947, do you still face this issue if you try the most recent build of the app? I no longer have the problem myself. Please let us know so the issue can be closed if it is resolved. https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc8

nowkillkennys, 2021-02-10:
will this work with he apple m1 chip?

VinnyCordeiro, 2021-02-10:
@nowkillkennys as @a2k-hanlon said, you could run from source by running it with Python. Far from ideal, I know, but better than nothing.

a2k-hanlon, 2021-03-17:
Closing due to lack of feedback/activity on the original issue.

closed 2021-03-17 by a2k-hanlon