kliment/Printrun #1021: Mac precompiled doesn't work on macOS 10.15 (Catalina).

drewshipley, 2019-10-09:
Downloaded the precompiled version for macOS from [here](http://kliment.kapsi.fi/printrun/) which is "Printrun-Mac-18Nov2017". When launched application produces the following error...

"Printrun-Mac-18Nov2017" needs to be updated. The developer of this app needs to update it to work with this version of macOS. Contact developer for more information."

fieldOfView, 2019-10-18:
Have you tried the 2.0.0 RC5?
https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc5

leonelhs, 2019-10-23:
Its very easy to build it from the sources just following the buildinstructions.txt 
Here is my build link: [Printrun-Mac-22Oct2019.zip](https://drive.google.com/open?id=1EM29GsKn4BgIRkGDUfsydpVr2sDctWuU)

closed 2020-03-31 by kliment