kliment/Printrun #1022: mac os x catolina

aryanveisi, 2019-10-11:
your app wont open in mac os x catolina becouse its 32 bit

androuino, 2019-10-14:
same thing here. I hope they release a new update soon.

fieldOfView, 2019-10-18:
Have you tried 2.0.0 rc 5:
https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc5

leonelhs, 2019-10-23:
Its very easy to build it from the sources just following the buildinstructions.txt 
Here is my build link: [Printrun-Mac-22Oct2019.zip](https://drive.google.com/open?id=1EM29GsKn4BgIRkGDUfsydpVr2sDctWuU)

georgelza, 2020-03-17:
have anyone been able to build for the new Catalina ? and maybe consider making available,
for those not aware, all previous version was 32bit it seems, MacOS Catalina only supports 64bit binaries, so we need someone to recompile/build for 64 bit only.
?
G

georgelza, 2020-03-17:
leonhs... can say thank you, got it opened on my machine.
G

georgelza, 2020-03-17:
may i ask, whats the difference between pronterface and print run-mac

SkipperDo, 2020-10-17:
I tried to compile it on my own machine (MBP 2019) and it works...
I can open pronterface by using command python pronterface.py

But im unable to build an app..
Get error messages while tring to make the app..
any suggestions?

Error Message: RuntimeError: Polyfit sanity test emitted a warning, most likely due to using a buggy Accelerate backend. If you compiled yourself, see site.cfg.example for information. Otherwise report this to the vendor that provided NumPy.
RankWarning: Polyfit may be poorly conditioned

volconst, 2020-10-20:
Try workarounds from here https://github.com/numpy/numpy/issues/15947
Install and configure openblas
or
pip3 install numpy==1.18.0
or
and uninstall numpy and
pip3 install --only-binary :all: numpy

SkipperDo, 2020-11-13:
I think there is an other failure while compiling the Sources....

Error Messages while Compiling:
Using legacy 'setup.py install' for wxPython, since package 'wheel' is not installed.
Using legacy 'setup.py install' for cairocffi, since package 'wheel' is not installed.
Using legacy 'setup.py install' for psutil, since package 'wheel' is not installed.

Could this maybe the issue?

volconst, 2020-11-14:
@SkipperDo , this does not look like an error, but just a warning/notice.
I would naively try 'pip3 install wheel'

a2k-hanlon, 2021-02-02:
@SkipperDo I had a lot of issues in the past with building the app on macOS, but with all the latest/prescribed dependencies and the source up to date I can build no problem. If you just want to use the program, please try the recent release https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc8. If you would like to get your local build working, I would suggest first starting fresh by pulling the latest source and following all the setup instructions, including creating a new virtual environment.

This issue in it's entirety is somewhat out of date so I will close it. However, if anyone runs into issues with the latest release/source, please feel free to open a new issue. Thanks!

> may i ask, whats the difference between pronterface and print run-mac

@georgelza, under release 2.0.0rc5, those 2 mac builds are the exact same, except the `Printrun-mac-2.0.0rc5.dmg` one is code-signed and the other is not. In any case, please use the most recent release instead.

closed 2021-02-02 by a2k-hanlon