kliment/Printrun #1023: Printcore Reading Temp

tarikcaliskan, 2019-10-29:
I have been working on controlling my 3D printer with print core. So I need to read temperatures. When I wrote like this, I got these outputs. Is my code wrong? Is it a bad function usage? How can I do this with printcore? 

Notes: I am working on Raspberry Pi 3 B+ with Raspbian Buster. My printer is working on Arduino Mega(2560) with Ramps 1.4 on Marlin. There is no other software to connected by serial on my printer.

```
from printrun.printcore import printcore
from printrun import gcoder
import sys
import time

p = printcore('/dev/ttyUSB'+ sys.argv[1], 250000)
while not p.online: time.sleep(0.1)
p.send_now("M105")
time.sleep(1)
p._listen()
```

```
pi@raspberrypi:/var/www/html/engine $ python3 send.py 0
WARNING:root:Memory-efficient GCoder implementation unavailable: cannot import name 'gcoder_line' from 'printrun' (/var/www/html/engine/printrun/__init__.py)
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 501, in read
    'device reports readiness to read but returned no data '
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/var/www/html/engine/printrun/printcore.py", line 271, in _readline
    line = self.printer.readline().decode('ascii')
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 509, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/var/www/html/engine/printrun/printcore.py", line 358, in _listen
    line = self._readline()
  File "/var/www/html/engine/printrun/printcore.py", line 292, in _readline
    if 'Bad file descriptor' in e.args[1]:
IndexError: tuple index out of range

```

Thanks, 
Tarık

volconst, 2019-12-18:
You should not call _listen(). It is called in its on thread by printcore class.
This works on Ubuntu 16.04

<pre>from printrun.printcore import printcore
import sys
import time

def temp_callback(a):
    print('temp_callback', a)

p = printcore('/dev/ttyUSB'+ sys.argv[1], 115200)
p.tempcb = temp_callback
while not p.online:
    time.sleep(0.1)
print('online')
p.send_now("M105")
time.sleep(1)
p.send_now("M105")
time.sleep(1)
p.send_now("M105")
time.sleep(1)
#disconnect to exit the listening thread and the main thread
p.disconnect()
</pre>

closed 2020-05-10 by tarikcaliskan

Bridge313, 2022-08-05:
Is there a way to save the returned temperature into a separate variable?

rockstorm101, 2022-08-06:
> Is there a way to save the returned temperature into a separate variable?

@Bridge313 please see #1221 for a first hint. You would need to write your own callback function `printcore.recvcb` or use a handler like is used on the plugin sample.

Bridge313, 2022-08-09:
Hello thanks for the reply. I will give it a shot, if I am still confused I will create a new issue post.