kliment/Printrun #1024: Pronterface issue

Casthian, 2019-11-02:
Hi,
I have Arduino mega with Ramps 1.4, I use Merlin frame (latest version) on it. My Z axis moving if I click on X home button but if I click on Z home it works too. Whats wrong? Can you help me?

tarikcaliskan, 2019-11-03:
Hi,
There might be something wrong with Marlin. Send `G28 X;` command to printer manually. If the printer homes Z before X axis again, something blocks to do it in firmware configuration. If it is not, problem is about Pronterface.

rockstorm101, 2020-05-09:
Hi @Casthian , I know the issue is quite old, but were you able to find the root cause for such an odd behaviour? Did you try the [suggested troubleshooting][1]?

[1]: https://github.com/kliment/Printrun/issues/1024#issuecomment-549151388

rockstorm101, 2020-12-12:
Closing due to lack of feedback. Please feel free to reopen if needed.

closed 2020-12-12 by rockstorm101