kliment/Printrun #1026: I cant connect my printer to prontoface

drefster, 2019-11-03:
I am getting an issue with my PID, where my hot end would heat up and cool down in an endless loop, not letting the print start. I have managed to Identify the issue after consulting discord at r/ender3, But I can't seem to find any answers about my new issue. To access the 1.1.5 motherboard I need a gcode console, which I decided to use prontoface for I don't have the budget to get simplify 3d. So anyway fast forward about half an hour and I have got prontoface installed, I hook up my computer to my ender 3 with a mini USB to USB connection and I get the "something has been connected to your computer" sound and my ender 3 display turns on very dimly. but when I go to prontoface I can't find any USB connections and only find a connection to a USB serial bus(which is connected to my display), So how do I fix this?

volconst, 2019-11-05:
Did you install the driver for the boards serial chip? it's probably CH340, check your board. See https://www.th3dstudio.com/knowledgebase/ch340-drivers-th3d-uno-creality-v1-1-4-board/
Try to power the printer as normal, instead of relying on USB power.

rockstorm101, 2020-12-12:
Closing due to lack of feedback. Please feel free to reopen if needed.

closed 2020-12-12 by rockstorm101