kliment/Printrun #1027: Issue connecting to a remote port

DemoN3579, 2019-11-22:
When I try to connect from Pronterface to my 3D printer through the network. An error occurs:
`Connecting...
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/home/dmitriy/tmp/Printrun/printrun/printcore.py", line 356, in _listen
    self._listen_until_online()
  File "/home/dmitriy/tmp/Printrun/printrun/printcore.py", line 319, in _listen_until_online
    self._send("M105")
  File "/home/dmitriy/tmp/Printrun/printrun/printcore.py", line 683, in _send
    self.printer.write((command + "\n").encode('ascii'))
TypeError: write() argument must be str, not bytes
`
The problem occurred after upgrade Ubuntu to 19.10 and it happens whether I use Pronterface from a package or from a repository

volconst, 2019-11-22:
Delete .encode('ascii') at line 683 Printrun/printrun/printcore.py. Maybe the serial library has changed.

DemoN3579, 2019-11-22:
It did not help. Line 683 now looks like this
` self.printer.write((command + "\n"))`
resulting error
`Connecting...
Can't write to printer (disconnected ?):
Traceback (most recent call last):
  File "/home/dmitriy/tmp/Printrun/printrun/printcore.py", line 683, in _send
    self.printer.write((command + "\n"))
io.UnsupportedOperation: not writable
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/home/dmitriy/tmp/Printrun/printrun/printcore.py", line 271, in _readline
    line = self.printer.readline().decode('ascii')
  File "/usr/lib/python3.7/socket.py", line 586, in readinto
    raise OSError("cannot read from timed out object")
OSError: cannot read from timed out object
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/home/dmitriy/tmp/Printrun/printrun/printcore.py", line 356, in _listen
    self._listen_until_online()
  File "/home/dmitriy/tmp/Printrun/printrun/printcore.py", line 325, in _listen_until_online
    line = self._readline()
  File "/home/dmitriy/tmp/Printrun/printrun/printcore.py", line 292, in _readline
    if 'Bad file descriptor' in e.args[1]:
IndexError: tuple index out of range`

Azertyaaaa, 2019-12-09:
Same problem here.

volconst, 2019-12-18:
Please test if [this branch](/volconst/Printrun/tree/socket-error) fixes the problem and if so I will create a pull request to integrate it officially

DemoN3579, 2019-12-28:
> Please test if [this branch](/volconst/Printrun/tree/socket-error) fixes the problem and if so I will create a pull request to integrate it officially

It works. In this branch the problem is resolved

volconst, 2019-12-29:
Thanks, created pull request #1030

commit 5f98a781 (2020-01-02): Merge pull request #1030 from volconst/socket-error

Fix #1027, reimplementing tcp printer communication
commit c6228e28 (2020-01-02): Fix #1027, reimplementing tcp printer communication
Python 3 does not support non blocking/timeoting readline with sockets.
Reimplement timeouting readline using selectors package
closed 2020-01-02 by kliment