kliment/Printrun #1029: Pronterface freezes while start up and doesnt responce

timants, 2019-12-15:
i have so many problems with pronterface  , when i start it up in the folder it start up and freezes if i click something i get this programma doesnt respond, (in dutch (reageert niet)) and i have to brute froce it sown becouse it will not recover,photo: https://gyazo.com/b78a76913d76b3c9982fd3f153aa0673
,
What i tryed , deleteing everything all maps in appdata and users in c etc, but thats not working having the same issue, restarting pc is not helping,  this pronterface version is from 2015 and 2017 , 

------------------
the newest pronterface gives more porblems becouse that will not even install or open at all. 
i get this error: https://gyazo.com/20424eea3ceac2be5dec4ac01b8254af

i really hope you can help me becouse i love using the program to test my printers with and send custom gcode to it , 

(i have no log becouse it didnt generate it i think )

thanks,
best regards, tim

volconst, 2020-01-06:
Which windows version are you using?
Try building as described in buildinstructions.txt, section "setup on windows:"

rockstorm101, 2020-12-12:
Closing due to lack of feedback. Please feel free to reopen if needed.

closed 2021-01-31 by rockstorm101