kliment/Printrun #1031: Connect wirelessly ?

QuitButton, 2020-01-04:
I've just added a wifi>serial port adaptor to my printer - https://github.com/luc-github/ESP3D, which is working nice. I was wondering if/how I can connect to this?

volconst, 2020-01-06:
If you have the printer's ip and port you should be able to connect by entering ip:port instead of serial device.

QuitButton, 2020-01-08:
I'm not sure what I'm doing wrong here, I enter 192.168.0.245:8888, yet when I try to connect it brings up the error:
[ERROR] Could not connect to 192 at baudrate 115200:
Serial error: could not open port '192': WindowsError(2, 'The system cannot find the file specified.')

volconst, 2020-01-08:
Are you running the latest master branch?

volconst, 2020-01-08:
Check if you have leading, trailing spaces in the port input box and delete them.

crazy72, 2020-01-09:
Wireless is not working properly. When you select SD Upload it starts printing instead to load the file on SD card. (I'm using MKS TFT35).
Anyone else experiencing the same ?