kliment/Printrun #1034: Ubuntu Pronterface won't start after UI change

Czary2, 2020-01-10:
Hi, I'm new to Pronterface and without many thinking, changed the UI to tabbed version. Now all I see in the console is :
_WARNING:root:RPC server bound on non-default port 7979
Traceback (most recent call last):
  File "/usr/bin/pronterface", line 66, in <module>
    app = PronterApp(False)
  File "/usr/lib/python2.7/dist-packages/printrun/pronterface.py", line 2296, in __init__
    self.mainwindow = PronterWindow(self)
  File "/usr/lib/python2.7/dist-packages/printrun/pronterface.py", line 220, in __init__
    self.reload_ui()
  File "/usr/lib/python2.7/dist-packages/printrun/pronterface.py", line 279, in reload_ui
    self.createTabbedGui()
  File "/usr/lib/python2.7/dist-packages/printrun/gui/__init__.py", line 155, in createTabbedGui
    add_extra_controls(extracontrols, self, page1panel2, controls_sizer.extra_buttons)
  File "/usr/lib/python2.7/dist-packages/printrun/gui/controls.py", line 299, in add_extra_controls
    add("tempgraph", root.graph, flag = wx.EXPAND | wx.ALL, border = 5)
  File "/usr/lib/python2.7/dist-packages/printrun/gui/controls.py", line 126, in add
    container.Add(widget, *args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk3/wx/_core.py", line 15931, in Add
    return _core_.GridBagSizer_Add(*args, **kwargs)
wx._core.PyAssertionError: C++ assertion "!CheckForIntersection(item)" failed at ../src/common/gbsizer.cpp(223) in Add(): An item is already at that position_

I've read the https://github.com/kliment/Printrun/issues/879 thread, but sadly cannot find config file in Ubuntu.

rockstorm101, 2020-01-10:
Hi @Czary2,

Have you tried searching for a file named "pronsolerc"? Could you please confirm nothing comes out when you type the following command on a terminal window?

`find ~ -name "*pronsolerc" 2>/dev/null`

Czary2, 2020-01-13:
Thanks a lot, got the file and edited to my demands.

closed 2020-01-13 by Czary2