kliment/Printrun #1035: Up and down cursor keys in G-Code viewer block after a single use

ianmackinnon, 2020-02-06:
Steps to reproduce:

1.   Open Pronterface with a G-Code file: `pronterface my-file.gcode`
2.   Navigate layers using “U” and “D” keys
3.   Navigate layers using up and down cursor keys
4.   Try to navigate layers again using “U” and “D” keys

Observed behaviour: “U” and “D” keys work repeatedly, but as soon as a cursor key is pressed, navigating layers becomes blocked and no further input is received from either cursor keys or “U”/“D” keys. Navigating layers becomes functional again after mouse input of some kind, eg. left-clicking in the viewer.

Some initial investigation in the code suggests that `keypress()` in `gcview.py` does not get called at all after the first cursor key input, until `wx.CallAfter(self.Refresh)` is triggered, eg. by mouse input.

I am running the lastest source from the master branch (commit `5f98a78`) on Ubuntu 19.10 using Python 3.7 and wxPython `4.0.7.post2`.

kliment, 2020-02-06:
I believe what's happening here is a focus issue - u and d are only handled by the gcode viewer, so nothing else can catch those events, but once you use an arrow key (which, to be clear, are not meant to switch layer in the embedded viewer, only the standalone viewer window) the event gets caught by the direction controls, and they gain focus. Clicking on the view gives the view focus again so u and d work again. I'm not entirely sure how to prevent this from happening without breaking the directional controls when those are in movement mode, suggestions welcome.

ianmackinnon, 2020-02-06:
Thanks for the quick and clear response.

I am totally new to this project, but in case it's useful I offer the following UI suggestions from a point of superficial understanding:

-   Deprecate up and down cursors in favour of only U and D in the G-code viewer, both embedded and standalone, and remove support in a subsequent release.
-   Deprecate up and down cursors in favour of only U and D in only the embedded G-code viewer, and remove support in a subsequent release 
-   Distinguish the focus more clearly in the application when the G-Code viewer is embedded (perhaps with some subtle colour hints) so it's visible to the user which area has focus, and prevent use of the cursor keys from changing focus between them.

Incidentally, how does one launch the standalone G-Code viewer? As I said, I'm new, but it wasn't obvious to me that there was one.

volconst, 2020-07-10:
@ianmackinnon The top gcode viewer window pops up when you double click the embedded viewer

volconst, 2020-09-16:
@ianmackinnon can you test whether the new rc7 version fixes the problem?