kliment/Printrun #1036: [2.0.0rc5] error in Readme.md

sk8nfool, 2020-02-21:
There are 2 problems in the Readme.md file. The first is an ambiguity about where commands are entered. In the description for using macros there is a "PC>" mark the I assume means P<something>C<onsole> which I eventually figured out indicated the text box next to the Send button in pronterface. I haven't tried the other commands in the suite so I can't comment on them. This needs some clarification.

The second problem is the definition of the "move_down" macro. It currently says
```
  G91
  G1 Z-{0}
  G92
```
I believe the G92 should be G90. The G91 puts the printer into relative mode and a G90 puts it back in absolute mode. The G92 with out arguments does nothing that I can figure out.

DivingDuck, 2020-04-11:
Regarding [G92], Set position. 
Per definition it will set all axis to 0 w/o arguments.

sk8nfool, 2020-04-12:
Actually it doesn't. The Marlin description for G92 at https://marlinfw.org/docs/gcode/G092.html just says it _can_ be used to set the coordinates to something. It doesn't explicitly state that the default is to set any particular value. And the description for G92.1 is ambiguous at best given the broken English.

DivingDuck, 2020-04-12:
Maybe not in Marlin. 
G92 is a well known and documented (and often used) code in CNC world since long time. That's from where it's coming from originally. 
Sometimes I'm thinking that the Marlin folks want to break with given codes and they don't care about other areas anymore. Hopefully not. Well, in the end it will cost only money of equipment if they ramming their tools in parts.
Some well known references:
https://reprap.org/wiki/G-code#G92:_Set_Position
https://www.cnccookbook.com/g54-g92-g52-work-offsets-cnc-g-code/

sk8nfool, 2020-04-12:
Those links have enough caviates that using G92 without explicitly specifying offsets is a bit of a crap shoot. It doesn't belong in the documentation for printrun or anything else without some kind of CYA statement. Something like "If your machine supports G92 without parameters ..."  Also it is unclear what happens to the E (extruder) axis. But specifying "G92 X0 Y0 Z0" accomplishes the same thing without all the ambiguity. 

Its also unclear what happens to the relative/absolute coordinate state.

commit 6221988c (2020-04-18): README: Fix example macro 3

Closes #1036
closed 2020-04-18 by rockstorm101