kliment/Printrun #1037: Connecting... taking forever on the Turnigy fabrikator 2 mini

TT-392, 2020-02-27:
I am trying to connect to my 3d printer on linux. I am talking about a turrnigy fabrikator 2 mini that I got second hand from someone. I can have successfully connected to it from repetier host on windows in the past.

So, I am running the aur version on arch: https://aur.archlinux.org/packages/printrun-git/
And when I select /dev/ttyACM0 and 115200 and click connect, it says Connecting... and nothing more and everything is still grayed out. I also tried ticking the debug communications box, and then I just get the message SENT: M105 over and over again.
I tried: echo G28 > /dev/ttyACM0 which does successfully home the printer. but tail -f /dev/ttyACM0 doesn't seem to return anything.

rockstorm101, 2020-04-18:
Hi @TT-392, I know the issue is quite old, but were you able to solve this problem? Did you try to connect to the printer, through the same port, using any other software (i.e.. Arduino)?

TT-392, 2020-04-30:
I sadly did not manage to solve the problem. As far as I know the only program I am able to connect to it with is repetier host on windows. I tried evertying from arduino to minicom to screen at the same baud rate as is used in repetier (115200) (repetier host on linux doesn't work either).
I haven't tried a whole lot on windows yet (mainly because it is a bit of a pain to boot into that os).

volconst, 2020-04-30:
Sending M105 is requesting temperature, which should happen after successful connect (this is a  personal opinion, not based on the assumptions of the code). Can you see in the log if the printer responds to the connect with a greeting message ("start", settings, version).
Try to push the reset button on the controller board a few times after pressing connect button.
Try to connect through pronsole, and also invoke "reset" command a few times after "connect".
Search in the connect/serial options, there was some flag (DTR if I remember) which requests restart of controller during connect.