kliment/Printrun #1039: pronterface not executing on rPi

euan2e5, 2020-03-31:
I've installed pronterface on my raspberry pi using this [guide](https://www.instructables.com/id/How-to-Control-a-3D-Printer-Via-Raspbery-Pi2/). When i run the pronterface.py file as suggested by the guide nothing happens. This may be due to the fact that i'm running raspbian strecth lite. Can somebody help me remedy the situation ?

kliment, 2020-03-31:
I think the issue is that that guide is for an older version of pronterface, which used python2. The current dependencies all use python3. You should check if printrun is available as a package on raspbian - maybe you can "sudo apt install printrun" if it is. If not, follow the instructions for "running from source" in the readme. I don't know if a prebuilt wxpython is available for that platform, if not you might have to build it locally (which sadly takes several hours on the rpi).

volconst, 2020-06-14:
@euan2e5 raspbian stretch lite is console only, no GUI. You can run ./pronsole.py