kliment/Printrun #1041: [2.0.0rc5] Can't update build dimensions.

DivingDuck, 2020-04-02:
Hi there. It seems that reading from config file (or updating/changing via Edit Settings) isn't working like expected. 
The build plate size shown in the GUI won't be resized to the values from the config file. Please see attached picture. Any suggestions? 
System Windows 10, Printrun compiled or run from actual source code (from today) via Visual Studio 2019
Best regards, DD

[image]

Talha909, 2020-04-10:
how you run this. When ever i try run i got error message. failed to execute

DivingDuck, 2020-04-10:
Actually from source via VS 2019 as I did some additional modifications for my printer.

DivingDuck, 2020-04-12:
@Talha909, I forgot to mention that you can compile your own version (At least for windows. This is what I did). Follow the instructions from file buildinstructions.txt and then run the batch file release_windows.bat. You will find two additional directories called [build] and [dist]. First one will be used for compiling and second is for the exe file. 

Remark:
You need to make a little adjustment in the batch file as that was written for Python 2.7. Change the line to your actual python 3 installation or if you have set up a environment path you can change from
 **d:\python27\python.exe** setup_win.py py2exe -v 
to 
 py setup_win.py py2exe -v
or if there is more than one python versions installed choose the one you like to use e.g.
py -3.7 setup_win.py py2exe -v

And maybe you need to install as well py2exe.

Talha909, 2020-04-13:
> @Talha909, I forgot to mention that you can compile your own version (At least for windows. This is what I did). Follow the instructions from file buildinstructions.txt and then run the batch file [...]

can you send me you exe file or prontoface. Its easy for me. my email is robotics459@gmail.com

DivingDuck, 2020-06-09:
This seems to be solved with fix [#1054 ](https://github.com/kliment/Printrun/pull/1063). Thanks.

volconst, 2020-06-10:
@Talha909 Instead of trying to get *.exe which works, just install python and double click pronterface.py or in the cmd.exe type pronterface.py to see error messages if any

closed 2020-07-03 by volconst