kliment/Printrun #1043: Windows & OSX binaries 5 years old

boelle, 2020-04-17:
the win bin is dated 5 years old, any reason that it has not been updated when the git is updated less than a day ago?

site: https://www.pronterface.com/ 
direct link: http://kliment.kapsi.fi/printrun/

kliment, 2020-04-18:
Please use this instead https://github.com/kliment/Printrun/releases

boelle, 2020-04-18:
where is the windows binary?

boelle, 2020-04-18:
oh i see it
but still 2 years old

boelle, 2020-04-18:
2.0.0 rc5

[image]

and that is before it opens the main window

DivingDuck, 2020-04-18:
Yea, that's the one that is crashing with windows 10. You need to compile it new.

boelle, 2020-04-18:
i dont know how, just a "dumb" windows user

boelle, 2020-04-18:
well 1.6.0 does the trick so far

DivingDuck, 2020-04-18:
Well, I cant attach it here, it's too big as attachment. So you need to wait for an update from the owner of this project.

rockstorm101, 2020-04-18:
> Yea, that's the one that is crashing with windows 10. You need to compile it new.

Hi @DivingDuck, do you happen to know what's changed in Windows that makes current binary useless?

DivingDuck, 2020-04-18:
Not really to be honest. I was running before v1.6 and v2.0RC2 w/o this problem. But RC5 never run on my PC with Windows 10.  
Mid last year I setup a new Python virtual environment in VS 2019 from where it build correct and, as I remember,  it took me a while to get this running.

DivingDuck, 2020-04-18:
@kliment, if you want, I can send you a compiled version as I did it already for other too.

DivingDuck, 2020-04-19:
@rockstorm101 . Your question let me not sleeping :) 
... anyway I had to figure out this for me too as with building a new environment I had the same problem with a crashing windows application's.

Here is the solution I found so fare: 

It turned out that the actual pyinstaller (v3.6) seems to be the root cause of this error. For more information you can follow a discussion about this issue here:
https://github.com/pyinstaller/pyinstaller/issues/2137

To solve the problem uninstall the actual pyinstaller and install instead the developer version for the virtual environment:
pip uninstall pyinstaller
pip install https://github.com/pyinstaller/pyinstaller/archive/develop.zip

There is also some confusion potential for "how to build a windows environment" for running and building a windows application. For most users the given "suggestions" at all the different places in files, wiki and readme's will lead to a point where nothing will work. Maybe a good point to clean up too.

rockstorm101, 2020-04-19:
Thanks a lot for sharing you findings, let's hope that pyinstaller is fixed soon, one cannot rely on development versions for too long.

Please feel free to submit a PR with the correct method for building on Windows. I believe this could save a lot of headaches to future Printrun users.

MichaelDu9226, 2020-06-06:
@DivingDuck Please share the correct method for building on Windows,thank you

DivingDuck, 2020-06-06:
I use VS2019 as programming environment and wrote me a little help instruction how I did my windows build. It took me too a while to find a way getting it running. Not perfect but maybe helpful for you too.
[https://github.com/DivingDuck/Printrun/wiki/How-to-setup-Printrun-with-Visual-Studio-2019](https://github.com/DivingDuck/Printrun/wiki/How-to-setup-Printrun-with-Visual-Studio-2019)

rockstorm101, 2020-12-12:
Hi @boelle and @DivingDuck, could you please confirm the latest version [2.0.0rc7](https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc7) works as expected? If so, would you consider this issue closed?

DivingDuck, 2020-12-13:
Hi @rockstorm101, the compiling issues should be gone since quite some time if you use the modified batch file. I just update my rep

Regarding RC7, there was an error what was solved from @volconst later. 
Reference: https://github.com/kliment/Printrun/pull/1069#issuecomment-692127342 and https://github.com/kliment/Printrun/issues/1111
It is maybe a good idea to make a new RC

For the main point in this issue here, there is an updated batch file in my repository (branch Gauges) called release_windows.bat that do more or less everything for compiling the windows version local. Maybe we should update the existing with my version.
https://github.com/DivingDuck/Printrun/blob/Gauges/release_windows.bat

I guess, you can close this issue.

rockstorm101, 2020-12-14:
> Hi @rockstorm101, the compiling issues should be gone since quite some time if you use the modified batch file. I just update my rep
> 
> Regarding RC7, there was an error what was solved from @volconst later.
> Reference: [#1069 (comment)](https://github.com/kliment/Printrun/pull/1069#issuecomment-692127342) and #1111
> It is maybe a good idea to make a new RC

Thanks, I understand the Windows executable then works fine but it doesn't include the new layer change algorithm. I'll close this issue on that basis then.

> For the main point in this issue here, there is an updated batch file in my repository (branch Gauges) called release_windows.bat that do more or less everything for compiling the windows version [...]

Good stuff, I would submit your proposal on a PR so it can be discussed in a more appropriate forum.

closed 2020-12-14 by rockstorm101