kliment/Printrun #1046: Printer stops when computer goes to power saving mode

a2k-hanlon, 2020-04-19:
Running the 2.0.0rc5 binary on macOS 10.15.4, the following error occurs as soon as a print is started (comes up right after "Print started at <time>"):
```
Traceback (most recent call last):
  File "printrun/pronsole.py", line 1219, in startcb
  File "printrun/power/__init__.py", line 130, in powerset_print_start
  File "printrun/power/osx.py", line 75, in inhibit_sleep_osx
  File "printrun/power/osx.py", line 52, in AssertionCreateWithName
  File "printrun/power/osx.py", line 44, in StringToCFString
ValueError: Expecting byte string of length 1, got a 'str'
```
When the computer enters a power saving mode, it stops communicating with the printer, and the printer stalls. As soon as I wake the computer, it starts up again, but the print is a goner.

I've had a similar issue occur in the past using earlier versions of macOS (eg. 10.13.x), but found I could avoid the issue by changing the computer's energy saving settings to inhibit sleep and hard disk sleep, as well as checking the "Prevent App Nap" in the application's Get Info dialog (that's no longer an option in macOS Catalina).

a2k-hanlon, 2020-05-04:
When running Printrun from source on my Mac (with the latest code, commit b517202), this issue does not occur. The print completes successfully without stalling.

I'm going to try packaging a build on my own machine and running that to check for this issue next.

rockstorm101, 2020-05-05:
Good news, what's changed from 2.0.0rc5 that makes this work properly now?

volconst, 2020-05-05:
@rockstorm101  https://github.com/kliment/Printrun/pull/937

a2k-hanlon, 2020-05-05:
Thanks @volconst, that's definitely it. My apologies, I didn't realize this was a duplicate of #946 too.

As for trying a packaged version of the program, I packaged the app according to the instructions in buildinstructions.txt but it didn't run. Going on the error messages and pyinstaller warnings, I changed a few things and got it to run, with several problems. The original issue at hand here didn't come up once I got the packaged app (sort of) running though, so that's good at least. The problems that present themselves when I packaged the program are unrelated I'm pretty sure.

I'll close this issue, as it's already been solved by the PR @volconst shared.

closed 2020-05-05 by a2k-hanlon