kliment/Printrun #1050: Missing bits on the Armenian translation files

rockstorm101, 2020-04-27:
Hi @Avag-Sayan, thanks a lot for helping with translating. I've checked the Armenian translation files and noticed the following on `plater.po` header fields:
 * 'Project-Id-Version' is at the default value
 * 'POT-Creation-Date' is duplicated
 * 'PO-Revision-Date' is missing

Also the description and copyright information is missing on both files. Could you please submit a PR with the fixes?

closed 2020-05-08 by rockstorm101

Avag-Sayan, 2021-04-11:
Hello, can you add the mo files for Armenian lang: plater.mo and pronterface.mo? I cannot because the file attachement is disabled.

rockstorm101, 2021-04-16:
Hi @Avag-Sayan, sorry it took me a while to reply. I'm no expert on this but it seems that '.mo' files are simply generated from '.po' files?

[Producing Binary MO Files](https://www.gnu.org/software/gettext/manual/html_node/Binaries.html#Binaries)

Avag-Sayan, 2021-04-19:
I can create the .mo file, but I cannot upload it.

-- 
Հարգանքով՝

Ավագ Սայան

___________________

Avag Sayan
*Technical & Methodology Specialist*
UATE.org
Hakob Hakobyan Str.,
3 Building, 3rd Floor
Yerevan, Armenia
Tel.:   (+374) 11 54 88 81 #16
Mob.: (+374) 55 28 81 44

DivingDuck, 2021-04-19:
You can attach the file here. I'm sure we can integrate the two language files when you have finish your translation.

Avag-Sayan, 2021-04-19:
Thanks...

-- 
Հարգանքով՝

Ավագ Սայան

___________________

Avag Sayan
*Technical & Methodology Specialist*
UATE.org
Hakob Hakobyan Str.,
3 Building, 3rd Floor
Yerevan, Armenia
Tel.:   (+374) 11 54 88 81 #16
Mob.: (+374) 55 28 81 44

commit 69113f9f (2020-05-08): Fix Armenian translation files

Fixes #1050.