kliment/Printrun #1051: Horizontal alignment flag-related wxPython errrors at startup

a2k-hanlon, 2020-05-04:
Running Printrun from source with the latest source code (commit b517202), the program hangs while starting up with an error such as the following:

```
Traceback (most recent call last):
  File "pronterface.py", line 62, in <module>
    app = PronterApp(False)
  File "/Users/Andrew/code/printrun/printrun/pronterface.py", line 2397, in __init__
    self.mainwindow = PronterWindow(self)
  File "/Users/Andrew/code/printrun/printrun/pronterface.py", line 219, in __init__
    self.reload_ui()
  File "/Users/Andrew/code/printrun/printrun/pronterface.py", line 296, in reload_ui
    self.settings.controlsmode == "Mini")
  File "/Users/Andrew/code/printrun/printrun/gui/__init__.py", line 233, in createGui
    controls_sizer = ControlsSizer(self, controls_panel, mini_mode = mini)
  File "/Users/Andrew/code/printrun/printrun/gui/controls.py", line 390, in __init__
    else: self.make_standard(root, parentpanel, standalone_mode)
  File "/Users/Andrew/code/printrun/printrun/gui/controls.py", line 423, in make_standard
    llts.Add(wx.StaticText(lltspanel, -1, _("XY:")), flag = wx.ALIGN_RIGHT | wx.ALIGN_CENTER_VERTICAL)
wx._core.wxAssertionError: C++ assertion "!(flags & wxALIGN_RIGHT)" failed at /Users/robind/projects/bb2/dist-osx-py37/build/ext/wxWidgets/src/common/sizer.cpp(2098) in DoInsert(): Horizontal alignment flags are ignored in horizontal sizers
```

I'm running Python 3.7.7 with wxPython 4.1.0 in macOS 10.15.4.

There are 7 places in the code that trigger this error. If I remove the offending flags wx.ALIGN_CENTER_HORIZONTAL or wx.ALIGN_RIGHT, the program will run successfully and I don't think anything looks off as far as the positioning of GUI elements.

I can commit these changes if that would be appreciated. If so, could someone kindly tell me how to proceed (branch and pull request, or otherwise)? I haven't contributed to this project before.

rockstorm101, 2020-05-05:
Please feel free to submit a PR against the master branch. Thanks a lot for contributing. I would only suggest to thoroughly investigate the issue. We wouldn't want to trigger a silent failure somewhere else.

a2k-hanlon, 2020-05-05:
No worries, I'm happy to help if I can! PR submitted. As for thoroughly investigating the issue, I'm afraid I don't think I have the familiarity with wxPython, let alone python, to do so. I'm really just fiddling with things rather naively in order to get the program working for myself!
I thought this might be related but again I don't really know: https://forums.wxwidgets.org/viewtopic.php?t=42182

commit 6dfdeec9 (2020-05-05): Merge pull request #1052 from a2k-hanlon/fix_flag_errors

Removed error-causing wxPython horizontal alignment flags. Closes: #1051.
closed 2020-05-05 by rockstorm101