kliment/Printrun #1058: Pyserial problem with M503

DivingDuck, 2020-05-15:
There seems to be an communication problem with pyserial and printrun regarding Unicode characters.
After running command M503 in console window the reported values are not shown completely and the serial communication is broken. There is a problem with some Unicode values in combination with smoothieware controllers. I run the old printrun 1.x version and there is the report complete. The problem character is ' ² ' or 'Â²' (and I guess for ' ³ ' too)

In attached file you can see where the problem start:

**;E acceleration mm/secÂ²:**

The left window shows the console window from v2.x and the right one from v1.x
[image]
.

Printrun2RC5 (actual source code from today)
Printrun 1.6.0 (for comparison)
OS: Windows 10 (64bit)

rockstorm101, 2020-05-16:
Hi @DivingDuck I cannot reproduce this on my machine. Could you please check whether changing this line solves the issue:

```
--- a/printrun/printcore.py
+++ b/printrun/printcore.py
@@ -334,7 +334,7 @@ class printcore():
             if line_bytes is PR_EOF:
                 self.logError(_("Can't read from printer (disconnected?). line_bytes is None"))
                 return PR_EOF
-            line = line_bytes.decode('ascii')
+            line = line_bytes.decode('utf-8')
 
             if len(line) > 1:
                 self.log.append(line)
```

DivingDuck, 2020-05-16:
Hi @rockstorm101, 
This works perfect. Thank you very much.

[image]

commit 1423eb46 (2020-05-16): p/printcore: Decode serial as UTF-8

Fixes #1058.
closed 2020-05-16 by rockstorm101